realsenseai/librealsense #1190: NUC7i5BNB vs RS300, R200, D435, ZR300

seath49, 2018-02-17:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 /  SR300 / ZR300 / D435 } | 
| Firmware Version                   | (3.21) | 
| Operating System & Version |   {Linux (OpenSuSe 42.3)  | 
| Kernel Version (Linux Only)    |  (4.1, 4.4, 4.9, 4.15 Suse and Vanilla)                                         | 
| Platform                                 | Intel NUC7i5BNB  bios v60|
| SDK Version                            |  { 2.10.0, 1.12}                          | 

Hi, 
I am trying to run realsense cameras on a NUC7i5BNB. I tried various kernel versions (4.1, 4.4, 4.9 and 4.15), both SuSe and Vanilla. Some of the kernels were patched and tested on other machines
where the cameras were working correctly. 

On NUC7i5BNB, no IR or depth streams work. 
## dmesg: ##

[  112.077457] Linux video capture interface: v2.00
[  112.092054] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[  112.094674] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3.1/2-3.1.2/2-3.1.2.1/2-3.1.2.1:1.0/input/input11
[  112.167868] uvcvideo: Unknown video format 435a4e49-b602-480f-978c-e4e88ae89b89
[  112.168797] uvcvideo: Unknown video format 52494150-8536-4841-b6bf-8fc6ffb083a8
[  112.169797] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) Camera SR300 (8086:0aa5)
[  112.172060] uvcvideo: Unable to create debugfs 2-5 directory.
[  112.173000] input: Intel(R) RealSense(TM) Camera S as /devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3.1/2-3.1.2/2-3.1.2.1/2-3.1.2.1:1.2/input/input12
[  112.174380] usbcore: registered new interface driver uvcvideo
[  112.175640] USB Video Class driver (1.1.1)

and when I try to open the streams I get a v4l crash on Unknown pixel format:
## dmesg ##

[  118.633681] WARNING: CPU: 3 PID: 2402 at drivers/media/v4l2-core/v4l2-ioctl.c:1297 v4l_enum_fmt+0xefe/0x1290 [videodev]
[  118.635397] Unknown pixelformat 0x00000000
[  118.637113] Modules linked in: uvcvideo(E) videobuf2_vmalloc(E) videobuf2_memops(E) videobuf2_v4l2(E) videobuf2_core(E) videodev(E) af_packet(E) ext4(E) snd_hda_codec_hdmi(E) crc16(E) jbd2(E) mbcache(E) input_leds(E) led_class(E) snd_hda_codec_realtek(E) snd_hda_codec_generic(E) hid_generic(E) usbhid(E) battery(E) thermal(E) snd_hda_intel(E) snd_hda_codec(E) coretemp(E) snd_hwdep(E) crct10dif_pclmul(E) iTCO_wdt(E) crc32_pclmul(E) snd_hda_core(E) iTCO_vendor_support(E) aesni_intel(E) snd_pcm(E) aes_x86_64(E) glue_helper(E) lrw(E) snd_timer(E) gf128mul(E) e1000e(E) ablk_helper(E) snd(E) i2c_i801(E) ptp(E) cryptd(E) mei_me(E) soundcore(E) pps_core(E) i2c_smbus(E) pcspkr(E) shpchp(E) mei(E) wmi(E) xfs(E) libcrc32c(E) sd_mod(E) i915(E) i2c_algo_bit(E) xhci_pci(E) drm_kms_helper(E) syscopyarea(E) xhci_hcd(E)
[  118.637147]  ahci(E)
[  118.637147]  sysfillrect(E)
[  118.637148]  sysimgblt(E)
[  118.637148]  libahci(E)
[  118.637148]  fb_sys_fops(E)
[  118.637149]  usbcore(E)
[  118.637149]  crc32c_intel(E)
[  118.637150]  libata(E)
[  118.637150]  usb_common(E)
[  118.637150]  drm(E)
[  118.637151]  video(E)
[  118.637151]  button(E)
[  118.637151]  sg(E)
[  118.637152]  scsi_mod(E)
[  118.637152]  autofs4(E)
[  118.645066] CPU: 3 PID: 2402 Comm: realsense-viewe Tainted: G            E   4.9.35-rt25-CR0-rt+ #11
[  118.645066] Hardware name: Intel Corporation NUC7i5BNK/NUC7i5BNB, BIOS BNKBL357.86A.0060.2017.1214.2013 12/14/2017
[  118.645069]  0000000000000000 ffffffff812f6f7b ffffc9000322fc70 0000000000000000
[  118.645070]  ffffffff8106c67f ffffc9000322fdc0 ffffc9000322fcc0 0000000000000000
[  118.645071]  ffff880456765240 0000000000000000 ffffffffa0711160 ffffffff8106c6ef
[  118.645072] Call Trace:
[  118.645076]  [<ffffffff812f6f7b>] ? dump_stack+0x5c/0x81
[  118.645078]  [<ffffffff8106c67f>] ? __warn+0xdf/0x100
[  118.645080]  [<ffffffff8106c6ef>] ? warn_slowpath_fmt+0x4f/0x60
[  118.645082]  [<ffffffff811cb5f6>] ? __kmalloc+0x26/0x200
[  118.645084]  [<ffffffff812ffbb7>] ? strlcpy+0x37/0x50
[  118.645090]  [<ffffffffa06fa88e>] ? v4l_enum_fmt+0xefe/0x1290 [videodev]
[  118.645093]  [<ffffffffa06fae72>] ? __video_do_ioctl+0x252/0x2d0 [videodev]
[  118.645095]  [<ffffffff811efa03>] ? path_openat+0x5f3/0x12b0
[  118.645098]  [<ffffffffa06fac20>] ? v4l_enum_fmt+0x1290/0x1290 [videodev]
[  118.645102]  [<ffffffffa06fd38c>] ? video_usercopy+0x1dc/0x510 [videodev]
[  118.645105]  [<ffffffffa06f65b5>] ? v4l2_ioctl+0x95/0xb0 [videodev]
[  118.645107]  [<ffffffff811f4f2f>] ? do_vfs_ioctl+0x8f/0x5d0
[  118.645108]  [<ffffffff811fff6b>] ? __alloc_fd+0xab/0x160
[  118.645110]  [<ffffffff811ff8d0>] ? __fget+0x70/0xb0
[  118.645111]  [<ffffffff811f54e4>] ? SyS_ioctl+0x74/0x80
[  118.645113]  [<ffffffff815aadf7>] ? entry_SYSCALL_64_fastpath+0x1a/0xa9
[  118.645114] ---[ end trace 0000000000000002 ]---
[... 158 lines cut ...]
	type                    : INTEGER
	name                    : "Sharpness"
	minimum                 : 0
	maximum                 : 100
	step                    : 1
	default_value           : 50
	flags                   : 0
    VIDIOC_QUERYCTRL(BASE+28)
	id                      : 9963804
	type                    : INTEGER
	name                    : "Backlight Compensation"
	minimum                 : 0
	maximum                 : 4
	step                    : 1
	default_value           : 0
	flags                   : 0

## realsense-viewer reports:
WARNING [140299746359040] (sensor.cpp:315) Unregistered Media formats : [ INZC PAIR YUYV ]; Supported: [ ]

Also, the camera (SR300) reports two formats which are nowhere in the patches or documentation
(fourcc: INZC and PAIR).
BIOS is at the newest version (v60), I tried the cameras plugged in directly or through a powered USB hub, same result. Any help would be appreciated.
Thanks, 
Seath

ev-mp, 2018-02-18:
Hello @seath49 ,
SR300 depth video formats were upstreamed into 4.12 kernel. Running Vanilla kernel with that version or later should allow you to receive both the depth and infrared data. 
There are additional utility formats that are not intended for upstream that are the warnings and kernel logs. (See a related [1137](https://github.com/IntelRealSense/librealsense/issues/1137) for details).

As for the logs attached -
Though you haven't specified the kernel version/distribution used when the logs there issued, I would assume it would the 4.15 kernel (otherwise you'd receive more than two `uvcvideo: Unknown video format` messages). As mentioned above - with 4.12+ streaming of SR300 Depth + IR should be available out of the box with all distributions.
In order to understand what went wrong please perform the following - 
 - Run `dmesg -wT`
- Reconnect SR300
- Run 'realsense-viewer' with Depth+IR configuration.
- Post the kernel and Application log for analysis

ev-mp, 2018-02-26:
Hi @seath49, do you still need assistance with this issue ?

RealSense-Customer-Engineering, 2018-03-13:
[Realsense Customer Engineering Team Comment] 
Hi @seath49,
Once the patch for INZC and PAIR formats applied, still see any issue?

seath49, 2018-03-15:
Hi,

@RealSense-Customer-Engineering 
correct me if I'm wrong, but there are no PAIR or INZC fourcc codes in the patches provided.
These codes missing are not reported on other computers where the cameras work flawlessly.

@ev-mp 
Sorry for the late reply. The message here is that there seems to be a USB problem on NUC7i5BNB. 
So I reached out for help, hoping somebody already went down this road. The missing formats are
reported by the camera, however, they are not reported missing on other computers with identical kernel. So I assume that the cameras don't play well with the NUC chipset?

Thanks, 
Seath

RealSense-Customer-Engineering, 2018-03-18:
[Realsense Customer Engineering Team Comment] 
Hi @seach49

As @ev-mp mentioned, you can refer to a related 1137 for details to patch INZC and PAIR (missing SR300 formats) in the related source codes to generate uvc drivers. Not sure how your other computers have no these two formats missing issue if no manual patches in kernel 4.14. You can run the command "v4l2-ctl --list-formats-ext -d /dev/video1" to show the result.

After the manual patches, you should see below format showing correctly.
----------------------------------------------
 Index       : 3
 Type        : Video Capture
 Pixel Format: 'INZC'
 Name        : IR with Confidence
 Size: Discrete 640x480
 Interval: Discrete 0.017s (60.000 fps)
 Interval: Discrete 0.033s (30.000 fps)
 Interval: Discrete 0.100s (10.000 fps)
 Size: Discrete 640x240
 Interval: Discrete 0.009s (110.000 fps)
 Interval: Discrete 0.017s (60.000 fps)
 Interval: Discrete 0.033s (30.000 fps)
 Interval: Discrete 0.100s (10.000 fps)

 Index       : 5
 Type        : Video Capture
 Pixel Format: 'PAIR'
 Name        : Custom IR
 Size: Discrete 640x480
 Interval: Discrete 0.017s (60.000 fps)
----------------------------------------------

RealSense-Customer-Engineering, 2018-07-09:
[Realsense Customer Engineering Team Comment]
Hi @seach49,

Still need any support for this topic? If not, will close this issue.

closed 2018-07-26 by dorodnic