realsenseai/librealsense #1191: How to record frames by R200 and save them on laptop?

prachi220, 2018-02-18:
I wish to take videos from Intel realsense R200 camera and save the frames in a directory. I have searched through all the possible questions and links but they direct me to [this](https://github.com/IntelRealSense/realsense_sdk_zr300) which I believe is for R300 and I don't know whether it would work on R200.
Currently I am using Ubuntu 17.10, installed the sdk from [here](https://github.com/IntelRealSense/librealsense/tree/v1.12.1). From this I get example codes inside build/examples. I assume I need to edit [cpp-capture.cpp](https://github.com/IntelRealSense/librealsense/blob/v1.12.1/examples/cpp-capture.cpp) in order to save the frames it otherwise shows in a window. I somehow need to save the rs::stream object in a specified filepath, but I can't seem to go about it.
Any kind of help would be appreciated :)

dorodnic, 2018-02-18:
Hi @prachi220 

The software offering that came with the R200 did not include any officially supported recording capabilities. The SDK for ZR300 might work for R200, but you should be aware that this was a very low-volume project and is not likely to receive future updates. 

If you don't mind trying something experimental, you could use our ROS-bag recorder for your problem. 
I drafted a demo that records 200 depth and color frames using librealsense 1.12.1 into a ROS-bag you can later access using RealSense Viewer or various SDK APIs: [rs-capture.cpp](https://github.com/dorodnic/librealsense/blob/record_lrs1/examples/capture/rs-capture.cpp)
It relies on another commit under-review (#1193) so make sure you fetch both if you want to give it a try. 

The way it works is by linking with 1.12.1 and 2.10.0 (these releases should be able to co-exist side-by-side), streaming from R200 into a [software device](https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device) that is recorded into a *.bag file. 

This is not an official solution (yet?) but rather something I hope could help you right now.

jlgarcia75, 2018-02-26:
Hello @prachi220,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

closed 2018-03-02 by jlgarcia75