realsenseai/librealsense #1192: How to get status of rs2::pipeline?

UnaNancyOwen, 2018-02-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| SDK Version                            | 2.10.0                                                     | 

### Issue Description
<code>rs2::pipeline</code> cannot get state whether it is started (running) or stopped.
Could you consider to provide API to get status of <code>rs2::pipeline</code>?

e.g.
```cpp
rs2::pipeline;
const bool state = pipeline.get_status(); // running : true, stoped : false
```

Do you have any good ideas?
Thanks,

zivsha, 2018-02-19:
You can use:
```cpp
auto p = pipeline.get_active_profile();
if(p)
    is_running = true;
```
I think that should work

UnaNancyOwen, 2018-02-19:
@zivsha Thanks!

closed 2018-02-19 by UnaNancyOwen