realsenseai/librealsense #1196: Is there a detailed SDK 2.0 manual?

tim-vdl, 2018-02-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   {Win (8.1/10)  | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.10.0 }                          | 

### Issue Description
Hi,

I'm very new to software development/GitHub/Visual Studio (data science background) and I want to write a program to capture both depth and color images, generate point clouds and use the vertices as input in other code I already developed.

I installed the SDK 2.0 on windows and all examples under "C:\Program Files (x86)\Intel RealSense SDK 2.0\samples" are working fine. I'm also able to use the pyrealsense2 package for python and can do all the things showed in the examples under ["librealsense/wrappers/python/examples"](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples) on Github. Now I want to write an other program that combines parts of these examples in python or C++. However, for both languages I'm stuck.

I've been searching both the GitHub page and intel website for documentation but could not find anything to get me further.

Question 1: is there a manual that shows how to start a new project in Visual Studio using C++? I want to make a new project similar to the examples under "C:\Program Files (x86)\Intel RealSense SDK 2.0\samples" but I don't know how to set everything up.

Question 2: is there a manual that gives an overview of all functionalities in the SDK? For example, I aligned depth and color images, but I want to generate point clouds from these images in real world coordinates. I don't know where to find information on how I could do this. Also, is there such a manual of all functionalities for using the SDK in python? 

Hopefully someone can help me!
Thanks in advance!

Kind regards,
Tim

gjmsoftware, 2018-02-19:
Hi @tim-vdl ,
I will try to lend a hand, but I'm not the most knowledgeable person on the new SDK or the platform you are using. Regardless, see below:

1. To assist in setting up a project with Visual Studio (especially if you are a beginner), I would suggest looking through the documentation for Visual Studio. See this [link](https://blogs.msdn.microsoft.com/vcblog/2017/04/21/getting-started-with-visual-studio-for-c-and-cpp-development/) for a good starting point. Once the project is setup, you will need to setup the proper references so that the SDK will work correct. I would suggest looking at the current C++ examples and see how it was done there... I'm working in C# and just had to have the proper 'realsense2.dll' and 'LibrealsenseWrapper.dll' referenced to my project.

2. To my knowledge, there is not a manual for the SDK 2.0. The best I can suggest is to look through the [wiki](https://github.com/IntelRealSense/librealsense/wiki) (link on the top taskbar of this Git page), read through the other examples on this repo, or cast a wider net by putting your questions on the [Intel Realsense Community](https://communities.intel.com/community/tech/realsense) forum. 

Hope this helps.

dorodnic, 2018-02-21:
Hi @tim-vdl 

To set up a new Visual Studio project:
1. Create new Visual C++ project
2. Go to View > Other Windows > Property Manager 
3. Click on "Add Existing Property Sheet" icon
4. Navigate to "C:\Program Files (x86)\Intel RealSense SDK 2.0"
5. Add `intel.realsense.props`
6. You can now add `#include <librealsense2/rs.h>` and start working

Please take a look at our [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples) on GitHub, these include step by step introduction for each sample.

closed 2018-02-27 by jlgarcia75