realsenseai/librealsense #1197: 3DSeg for SDK 2.0

cvasquez-github, 2018-02-19:
Hello, we are upgrading our RealSense projects to support the new D415 and D435 cameras, so we needed to switch from discontinued SDK R6 to SDK 2.0. The problem we are facing is that we used the included Backgrounds Segmentation functions a lot and they are not included anymore. The new SDK includes align functions to match depth and color streams pixels, to add depth info to color pixels, but doesn't return a good/clean segmented user stream like the previous SDK. Could you please port the previous 3DSeg() functions from R6 to the new SDK 2.0? Or could you please share the source code of your previous SDK 3D Segmentation functionalities so we can recreate them in our side? Thank you very much!

closed 2018-02-27 by jlgarcia75

cubicmoviesam, 2018-07-26:
Hi @cvasquez-github  did you come up with a work around or you had to go back to previous version?

cvasquez-github, 2018-07-26:
Hi @cubicmoviesam , nope,  no work around yet... the problem with the previous version is that it works only with SR300 and we need it to make it work on 415 and 435 cameras... I don't really understand why they are not including such a useful functionality in their SDK that was already available in the old version...

cubicmoviesam, 2018-07-26:
Thanks @cvasquez-github  I agree. I really need this functionality right now and I have the D cameras . Whats the major draw back from your experience if I just decide to go back and use the sr300, since it doesnt appear like they'll bring the functionality into the ver 2.0 any time soon?

cvasquez-github, 2018-07-26:
1- The quality/resolution of the depth frame in SR300 is low (640x480) so the quality of the background segmentation you can get would be lower than being able to perform the same thing using the new D4XX cameras (Up to 1280 x 720). 

2- The other issue is the supported short range (distance) so using SR300 cameras you can only get half body shot (about a meter), and we wanted to remove the background on longer distances for full-body shots (About 0.2m-10m in D4XX cameras). 

Other than that, it works fine. Way better than using the a D415 with the new SDK and the rs2::align object.

cubicmoviesam, 2018-07-26:
@cvasquez-github  Thank you so much. I appreciate that. I hope they include it soon , at least on the next version.  Thank again for your insight