realsenseai/librealsense #1198: RealSense d435 is not recognized (No device connected...)

agari1229, 2018-02-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | realsense D435 | 
| Firmware Version                   | | 
| Operating System & Version | Ubuntu 16.04 LTS| 
| Kernel Version (Linux Only)    |  4.13.0-32-generic | 
| Platform                                 | PC |

### Issue Description
<Describe your issue / question / feature request / etc..>
I followed [this link](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) . 
The result from  
`roslaunch realsense_ros_camera rs_camera.launch`
is 
`[ERROR] [1519064413.637674817]: No RealSense devices were found! Terminating RealSense Node...
[FATAL] [1519064413.833545927]: Failed to load nodelet '/camera/realsense_ros_camera` of type `realsense_ros_camera/RealSenseNodeFactory` to manager `realsense_ros_camera_manager'
[camera/realsense_ros_camera_manager-2] process has died [pid 4279, exit code 1, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense_ros_camera_manager __log:=/home/yuki-ag/.ros/log/86d4b45e-15a1-11e8-88b0-d4258b2da0d7/camera-realsense_ros_camera_manager-2.log].
log file: /home/yuki-ag/.ros/log/86d4b45e-15a1-11e8-88b0-d4258b2da0d7/camera-realsense_ros_camera_manager-2*.log
[camera/realsense_ros_camera-3] process has died [pid 4280, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense_ros_camera/RealSenseNodeFactory realsense_ros_camera_manager __name:=realsense_ros_camera __log:=/home/yuki-ag/.ros/log/86d4b45e-15a1-11e8-88b0-d4258b2da0d7/camera-realsense_ros_camera-3.log].
log file: /home/yuki-ag/.ros/log/86d4b45e-15a1-11e8-88b0-d4258b2da0d7/camera-realsense_ros_camera-3*.log`

And 
`modinfo uvcvideo | grep "version:"`
is
nothing. 

i can find realsense d435 when i run `lsusb` commando.

icarpis, 2018-02-22:
Hello @agari1229, could you please provide the `lsusb` output?
Have you managed to streaming using the Intel RealSense tools/examples such as `realsense-viewer`/`rs-capture`?

RealSense-Customer-Engineering, 2018-03-07:
[Realsense Customer Engineering Team Comment] 
Hello @agari1229,
Are you still having this issue?

If you are, please provide the lsusb output. Can you also tell me the output from rs-enumerate-devices?

dorodnic, 2018-03-19:
Hi @agari1229 
Feel free to open a new issue if you still having this problem.

closed 2018-03-19 by dorodnic