realsenseai/librealsense #1202: python wrapper full list of commands

GitwellAnyohub, 2018-02-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

GitwellAnyohub, 2018-02-20:
I'm trying to use the python wrapper to record and save images. I would like to know if there is a full list of the pyrealsense2 commands (either documentation or step by step instructions on what to click so I can get there).

zivsha, 2018-02-21:
Hi @GitwellAnyohub ,
The Python wrapper has almost the exact same API as the C++ API, we do not have any specific documentation for the python wrapper but we do have several examples for using it under the python wrapper folder. You can see a list of the C++ API functions [here](https://github.com/IntelRealSense/librealsense/tree/master/include). If you have any specific question please share.

GitwellAnyohub, 2018-02-22:
Awesome thanks. I'll take a look

closed 2018-03-02 by jlgarcia75