realsenseai/librealsense #1206: Corrupted color image after pipeline restart

taketwo, 2018-02-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 08.05.15.00 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.10.0                                         | 
| Platform                                 | Lenovo laptop  |
| SDK Version                            |  2.10.0                          | 

### Issue description

Stopping and immediately starting a pipeline again leads to artifacts in color (but not depth) image:

[image]

It looks like odd and even columns are swapped. Also if you pay attention, the first column is swapped with the last.

### How to reproduce

Add after [line 22 in rs-capture.cpp](https://github.com/IntelRealSense/librealsense/blob/51e766389f75731be6d37a4f435b5c6b172609bc/examples/capture/rs-capture.cpp#L22):

```cpp
pipe.stop();
pipe.start();
```

For me corruption does not happen in 100% runs, but in a vast majority.

dorodnic, 2018-02-21:
Hello @taketwo 
Thank you for reporting this issue - we have never encountered anything like it. 
Could you please:
1. Try to reproduce with `-DBUILD_WITH_OPENMP=false` CMake flag. If the issue disappears, it might be related to our multi-threaded color conversion code. 
2. Configure the color sensor to Y16 or YUY formats. You can do this in the Viewer or via code: 
```cpp
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_COLOR, RS2_FORMAT_YUYV);
pipe.start(cfg);
// And also modify example.hpp to display YUY as 16-bit grayscale
```
If the problem still reproduces, the next logical step would be to check `dmesg` for any USB related warnings.

taketwo, 2018-02-21:
Hi @dorodnic, thanks for the quick reply. It's my honor to be the first one with this issue :smile:

1. I can reproduce this problem with OpenMP disabled. (By the way, turning it off lead to linker errors for some of the example executables, which I resolved by adding `pthread` to the librealsense target. I will report this separately.)

2. I tried to configure the color sensor to Y16 format:

```cpp
cfg.enable_stream(RS2_STREAM_COLOR, -1, 640, 480, RS2_FORMAT_Y16, 30);
```

and display it using:

```cpp
case RS2_FORMAT_Y16: 
  glTexImage2D(GL_TEXTURE_2D, 0, GL_RGB, width, height, 0, GL_LUMINANCE, GL_UNSIGNED_SHORT, frame.get_data());
```

The result has same features: swapped odd/even columns, swapped first and last columns.

[image]

3. I run `dmesg -wH` in the background and started the capture example (modified version with pipeline restart). The following lines show up:

```
[ +14.807442] nvidia-modeset: Allocated GPU:0 (GPU-91b8ff83-0575-25c2-7949-c817f43f996e) @ PCI:0000:08:00.0
[  +0.000450] nvidia-modeset: Freed GPU:0 (GPU-91b8ff83-0575-25c2-7949-c817f43f996e) @ PCI:0000:08:00.0
[  +0.181645] uvcvideo: Non-zero status (-71) in video completion handler.
[  +0.010190] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
```

If I run unmodified capture example, then the following lines show up:

```
[Feb21 21:39] nvidia-modeset: Allocated GPU:0 (GPU-91b8ff83-0575-25c2-7949-c817f43f996e) @ PCI:0000:08:00.0
[  +0.000443] nvidia-modeset: Freed GPU:0 (GPU-91b8ff83-0575-25c2-7949-c817f43f996e) @ PCI:0000:08:00.0
[  +0.180959] uvcvideo: Non-zero status (-71) in video completion handler.
[  +0.005020] uvcvideo: Non-zero status (-71) in video completion handler.
```

Does this shed some light? Thank you!

dorodnic, 2018-02-22:
Thanks, 
We were planning to issue the fix for OpenMP=false later today. Do you get similar output for YUY? Also, just making sure, the platform is x86, correct? (YUY to Y16 conversion is also using SSE2 on x86. If this reproduces with pure YUY we can rule out the user-space part and look at the lower levels). 
Kernel warnings seem normal to me (not a good thing in general, but we can't get rid of these for now).

taketwo, 2018-02-22:
Yes, it is a x86_64 machine. Sorry, I'm not well versed in video formats. When you say YUV you refer to `RS2_FORMAT_YUYV`, right? Could you please advise which parameters I should use in the `glTexImage2D()` call to be able to display this format?

dorodnic, 2018-02-22:
With the following patch `rs-capture` should show smooth gray-scale image:
```diff
diff --git a/examples/capture/rs-capture.cpp b/examples/capture/rs-capture.cpp
index 90f10a3..c0a7934 100644
--- a/examples/capture/rs-capture.cpp
+++ b/examples/capture/rs-capture.cpp
@@ -17,9 +17,12 @@ int main(int argc, char * argv[]) try
     rs2::colorizer color_map;

     // Declare RealSense pipeline, encapsulating the actual device and sensors
+    rs2::config cfg;
+    cfg.enable_stream(RS2_STREAM_DEPTH);
+    cfg.enable_stream(RS2_STREAM_COLOR, RS2_FORMAT_YUYV);
     rs2::pipeline pipe;
     // Start streaming with default recommended configuration
-    pipe.start();
+    pipe.start(cfg);

     while(app) // Application still alive?
     {
diff --git a/examples/example.hpp b/examples/example.hpp
index 0af1d09..1d271e2 100644
--- a/examples/example.hpp
+++ b/examples/example.hpp
@@ -90,6 +90,9 @@ public:
         case RS2_FORMAT_Y8:
             glTexImage2D(GL_TEXTURE_2D, 0, GL_RGB, width, height, 0, GL_LUMINANCE, GL_UNSIGNED_BYTE, frame.get_data());             break;
+        case RS2_FORMAT_YUYV:
+            glTexImage2D(GL_TEXTURE_2D, 0, GL_RGB, width, height, 0, GL_LUMINANCE_ALPHA, GL_UNSIGNED_BYTE, frame.get_data());
```

taketwo, 2018-02-22:
Thank you. With this patch I get a smooth gray-scale image. When I then add a pipeline stop/start, exactly same artifacts as before show up.

dorodnic, 2018-02-23:
Hi @taketwo 
So far, your tests suggest a problem inside YUY to RGB/Y16 conversion [here](https://github.com/IntelRealSense/librealsense/blob/master/src/image.cpp#L234).
I ran some tests to try and reproduce this problem (allocating unaligned memory buffers and stressing start / stop) but wasn't able to get a reproduction. 
To make sure this is related to conversion code, you can `#undef __SSSE3__` just before the function and see if it helps.
Also, my colleague submitted a [PR](https://github.com/IntelRealSense/librealsense/pull/1209) fixing some issues related to color conversions and adding AVX version, you can give it a try.

taketwo, 2018-02-23:
Sorry, I could not follow why the conclusion is that the problem is with the conversion. As I wrote in my last comment, the artifacts are still visible when visualizing `YUYV` stream directly.

Anyway, I tried both disabling SSE3 and using generic unpacking code, as well as the AVX2 version, same results. Works fine if the pipeline was started a single time, but shows artifacts when pipeline was started, stopped, and started again.

dorodnic, 2018-02-23:
Sorry, I misunderstood you. The second call to start is using the same config? 
Do you see this artifact in the Viewer? Do you have a Windows PC you can check the device on?
I will ask around if the RGB sensor is known to produce this effect in some conditions. Also, perhaps @ev-mp / @icarpis have any thoughts on what could go wrong in `uvcvideo` or USB level? 
Usually, when the host gets a partial / corrupted frame we block it from reaching the application.

aangerma, 2018-02-28:
Hello @taketwo
I was able to reproduce the problem on Ubuntu 16.04 with D415. 
We will look into it ASAP and update this thread.

dorodnic, 2018-03-21:
We are still investigating this issue on the firmware level, a work-around seem to be to add sleep of 200ms between stop and start.

taketwo, 2018-03-21:
Thanks for the tip!

dorodnic, 2018-06-12:
Fix is scheduled to be released in firmware 5.9.13

RealSense-Customer-Engineering, 2018-06-27:
[Realsense Customer Engineering Team Comment]
@taketwo This issue is fixed by FW 5.9.13. Please download from the link below and verify.

https://downloadcenter.intel.com/download/27924/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t

taketwo, 2018-06-27:
Thanks. Unfortunately I won't have access to the camera in the near future. I will close this issue for now and reopen if I face this bug again at some point.

closed 2018-06-27 by taketwo