realsenseai/librealsense #1207: Using align with post-processing

jb455, 2018-02-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |   Win 10  | 
| Platform                                 | PC, C#.NET  |
| SDK Version                            |  2.10.0                        | 

Hi,

My app uses Align to map the depth image to colour, so I have a 1:1 relationship between the colour image and aligned depth image. I've been playing with the post-processing filters and the problem I'm facing is that running the depth image through the filters reduces its dimensions so I no longer have the 1:1 mapping.
I've tried simply dividing the depth data index by the ratio between the size of the two images (colour and aligned depth which has been processed) like this:
```
var length = colourFrame.Width * colourFrame.Height;
var scale = length / (float)(depthFrame.Width * depthFrame.Height);

for(int idx = 0; idx < length; idx++)
{ 
    float distance = depthData[(int)(idx / scale)] * depthScale;
    uint colour = colourData[idx];
}
```
But this mapping doesn't work; I get a mess of depth values which don't correspond to the colour image. Is it possible to get align and post-processing working together like this?

Thanks,
James

ev-mp, 2018-02-22:
Hello @jb455 . Several notes/clarifications:
1. The filter that is capable of changing depth dimensions is the Decimation (down-sample) filter which applies proportionally to width and height of the depth image. The decimation factor is 2's exponent in range of [0-4] :
0=> 2^0 => 1 => Output size (x/1, y/1)
1=> 2^1 => 2 => Output size (x/2, y/2)
...
4=> 2^4 => 16 => Output size (x/16, y/16)

2. Can you elaborate the processing flow, as it is not clear, for instance whether you apply the filters before or after align step.

3. We recommend to apply post-processing filters before aligning depth to color to reduce aliasing. This is the way it implemented in ''realsense-viewer''s pointcloud view. 

4. When recalculating coordinates x and y of should be scaled separately as the input data is two-dimensional matrix:
For example : given frame size [16X8] the index of coordinate (2,6)  is (2X16+6) = **38** (indexes are row-major,zero-based).
When down-sampling by factor of 2, the new frame size is [8X4], the new location (1,3) and the coordinate index is 8*1+3=**11**.  And the correlation between 38 and 11 is not straight-forward.
But then again, changing the processing blocks order (grabbing frame->decimation->align) would allow to achieve the same effect while preserving 1:1 mapping in the final stage.

jb455, 2018-02-22:
Hi, thanks for your reply.

Currently I do the post-processing after alignment. I tried doing it before, but the depthFrame element of FrameSet is read-only so I can't modify the depth image with the filters then pass it to align, and there's also no FrameSet constructor as far as I can see (either in the wrapper or the native api) so I can't create a new one with my colour image and processed depth image.

Can you point me to where this is done in realsense-viewer's code please? I searched through the source code but couldn't find it using align anywhere. 

I was using nested loops in x & y before, but found that was really slow and using a single loop (for 0 to width * height) was about 5-10x faster so I don't want to go back to nested loops as that'll make my app unusable. Just to try it out though, I implemented it as follows:
```
var xScale = depthFrame.Width / (float)colourFrame.Width;
var yScale = depthFrame.Height / (float)colourFrame.Height;

for (int y = 0; y < colourFrame.Height; y++)
{
    for (int x = 0; x < colourFrame.Width; x++)
    {
        float distance = depthData[(int)((x * xScale) + (y * yScale)*depthFrame.Width)] * depthScale;
        uint colour = colourData[x + y * colourFrame.Width];
    }
}
```
I still don't get the correct mapping (and it's now even slower, about 0.2fps). But, as you say, if it is possible to pass the processed depth image to align, I won't need to do this.

RealSense-Customer-Engineering, 2018-03-21:
[Realsense Customer Engineering Team Comment] 
Hi,
I've changed the align.cpp example to have post processing works before doing alignment.

I added code inside to let you make sure the image is processed as expect, original one after applying the decimation will reduce to half (from 1280*720 to 640*360) and then after apply the align process, it will become 640*480 align with the RGB image.

      //checking the size before align process, due to decmiation set to 2, it will redcue the size as 1/2
      float width = depth.as<rs2::video_frame>().get_width();
      float height = depth.as<rs2::video_frame>().get_height();

      auto color = data.get_color_frame();
      rs2::frameset combined = source.allocate_composite_frame({ depth, color });
      combined = align.process(combined);

      //checking the size after align process, the size align to color 640*480
      width = combined.get_depth_frame().as<rs2::video_frame>().get_width();
      height = combined.get_depth_frame().as<rs2::video_frame>().get_height();

carypai, 2018-03-21:
Didn't attach successfully, please refer to this one
[rs-align.zip](https://github.com/IntelRealSense/librealsense/files/1832024/rs-align.zip)

jb455, 2018-03-21:
Hi, thanks for the code example! Is it possible to do it using the C# wrapper currently? I can see rs2_allocate_composite_frame is linked in NativeMethods but there's no FrameSource class to use it with, and it appears to return an rs2_source object rather than a FrameSet (unless source is castable to frameset?).

RealSense-Customer-Engineering, 2018-05-07:
[Realsense Customer Engineering Team Comment]
@jb455
The issue is solved in 2.11.0 LibRS, please refer to the cs-tutorial-3-processing example inside the csharp wrapper folder.

closed 2018-05-14 by RealSense-Customer-Engineering