realsenseai/librealsense #1212: Accessing output trigger signal with D435

pavolbauer, 2018-02-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 5.9.2 | 
| Operating System & Version |   Win10 | 
| Platform                                 | PC |
| SDK Version                            |  2.0 (build 2.10)        | 

### Issue Description
Hi,
According to the RS400 documentation, the D435 camera exposes headers to an output trigger signal indicating when a depth frame is being captured.
We tried to read out pin 1 from the external sensors sync connector (GVSYNC0) while enabling the "Output Trigger" option in the RealSense viewer but we could only measure a very noisy signal containing shutter noise, and not a clear trigger signal.
Is this the right way to access the output trigger (externally) or should this be done differently?
Best regards
Pavol & Kevin

jlgarcia75, 2018-02-27:
Hello @pavolbauer,

The external trigger functionality has not been fully implemented yet. It will be addressed in a future firmware version sometime in Q2'2018.

Regards,
Jesus G.
Intel Customer Support

pavolbauer, 2018-02-27:
Hi Jesus,

Thank you for your answer. So far we will work with the Depth VSYNC signal (pin 5 of the connector) which seems to deliver a 5V flank after each depth frame has been captured.

Best,
Pavol & Kevin

closed 2018-02-27 by pavolbauer

LorenzMeier, 2018-05-20:
@jlgarcia75 Is this signal suitable in any way to synchronize multiple D435 to capture at the exact same time?

ffrog12, 2018-06-13:
Has this been released yet? We are coming towards the end of Q2 18, I am looking to use the GPIO to trigger the camera or trigger an external camera.

LorenzMeier, 2018-07-04:
@jlgarcia75 Do you have an update on this? Our understanding from discussions with the Intel business was that we should expect a release around this time. Thanks!

jlgarcia75, 2018-07-05:
Hello, the external trigger functionality has not been released yet. I'm sorry I do not have a definite timeline for this. 

Regards,
Jesus
Intel Customer Support

timprepscius, 2018-07-10:
@pavolbauer Where did you buy the connector to the d435?

Do you have a model number?

wamageed, 2018-09-27:
Is there any updates in terms of the timeline?

jlgarcia75, 2018-09-28:
Hi @wamageed,

This capability is currently supported on the D435 with the latest FW, v 5.10.3, found here: https://downloadcenter.intel.com/download/28076/Latest-Firmware-for-Intel-RealSense-D400-Product-Family.

Regards,
Jesus
Intel Customer Support

mzahana, 2018-09-30:
@jlgarcia75  Is there documentation on how to use this functionality ?

jlgarcia75, 2018-10-01:
Hi @mzahana,

You can find the documentation here: https://www.intel.com/content/www/us/en/support/articles/000028140/emerging-technologies/intel-realsense-technology.html.

Regards,
Jesus G.
Intel Customer Support

mzahana, 2018-10-01:
Thanks @jlgarcia75

Do you know if the ROS package that works with the D435 handle the timestamping correctly for the external triggering mode?

m0oz, 2018-10-02:
@mzahana No I don't think so. I am trying to do the same thing but it seems like you would have to adapt the realsense ROS wrapper to get this functionality.
To me it seems like the realsense ROS package is lacking behind in functionality in regards to what is released with librealsense

jlgarcia75, 2018-10-25:
@m0oz,

External triggering is not possible today with RealSense ROS but it is in development. No timeline yet for when it will be released.

Regards,
Jesus 
Intel Customer Support

patrickpoirier51, 2018-12-07:
@mzahana have you looked at this 
https://github.com/ethz-asl/realsense/pull/2/files