realsenseai/librealsense #1213: Sporadic errors when, only workaround is to physically reconnect camera

maciekd, 2018-02-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.08.09.00 | 
| Operating System & Version |   Ubuntu 16 | 
| Kernel Version (Linux Only)    | 4.4.0-116-generic | 
| Platform                                 | Intel NUC |
| SDK Version                            |  2.10.0 | 

Hi,

I'm running a system with a number of RealSense cameras. We've been using the R200 and are now migrating to the R400 series.

About 50% of the time after system boot the cameras do not work.

Sometimes I get:
rs2_create_device(info_list:0x7fb4f8566930, index:0): libusb_bulk_transfer(...) returned LIBUSB_ERROR_IO Last Error: Resource temporarily unavailable

And other times there's no reply from the cameras at all and I can see in the sys log:
usb 3-4: device not accepting address 10, error -71

Only way to solve the problem is to reconnect the camera (sometimes several times). Once properly connected the camera seems to be working fine.

I'm on a Ubuntu server system that has been upgraded from the old librealsense to the new one (for the new devices). 

I've tested R415 and R435 cameras. Several of each so it's not a defect device. We're on Intel NUCs, I've tried several of those too. I'm using the original USB cables. Tried with and without USB3 hub. With or without external power to the USB3 hub. I've tried using only one camera connected, same problem.

Any idea what could be wrong? Any ideas on what I should try?

sunshinerain93, 2018-02-23:
This issue will be fixed by new firmware version. Please keep tuned.

maciekd, 2018-02-23:
Thanks for letting me know.

We're having this issue around 50% of the time, does that sound plausible? Just want to double-check.

Any tips on possible workarounds to mitigate the problem at least slightly? We'd love to continue working on the sensors asap.

In what kind of timeframe can we expect the new firmware? Weeks? Months?

jlgarcia75, 2018-02-27:
Hello @maciekd,

The timeframe for the new firmware is weeks. Sorry I cannot get more granular than that.

Regards,
Jesus G.
Intel Customer Support

maciekd, 2018-03-08:
If others have this problem: a temp. workaround seems to be to connect the cameras AFTER the system has booted up.

neomanic, 2018-03-20:
Same issue here. I get both that they either don't appear at all, or they enumerate as USB2 as per #1181. For the moment, reconnecting works, but that won't fly for when we get these into production so I'm watching this issue. (Hardware is an UP Board Squared, 2xD435s through a powered USB3 hub.)

jfhamlin, 2018-04-11:
@jlgarcia75 Hi Jesus, is there any update on this? I've encountered the same issue.

RealSense-Customer-Engineering, 2018-04-13:
[Realsense Customer Engineering Team Comment] 
Hello @jfhamlin,

Still no update. Engineering is working on a fix.

Regards,
Jesus
Intel Customer Support

josephduchesne, 2018-04-21:
I have found that on my 7th gen nuc this problem never occurs on the USB C interface. I automated a test and performed several thousand reboot cycles across a variety of configurations. 

The USB C port always worked (bus 4 on ubuntu 16.04) with a single camera and the USB a ports did not (bus 2).

I tried a variety of hubs to try to get more than one camera to work reliably on boot. A few worked reliably while most did not.

I will update this post later with model numbers and device names.

My configuration is a 7th generation i3 NUC with 2 D435 cameras, latest firmware, trying to provide only depth streams.

On a 5 minute reboot cron (for testing) I have a start up job that checks for both devices (lsusb ) and if both did not come up, it logs the error and reboots. If they are present it lauches a modified copy of the ros nodelet that first tries to perform a reset on the devices (to avoid another issue where the depth stream doesn't work). It then runs with a 10 second watchdog timer. If no depth frames arrive in 10 seconds it logs the error and restarts the entire nodelet  family for that camera.

I also have a hardware cold boot test rig (relay+arduino 15 seconds off, 5 min on) but found no significant difference in errors between cold boot and a reboot.

josephduchesne, 2018-04-23:
On an entirely different motherboard (6th gen intel i7 CPU 6700TE, Asrock IMB-195), I am seeing a very regular pattern of this sort with 2x D435s plugged in via the single USB3.0 bus.  Latest firmware (5.9.9.2).

After a reboot cycle: 
- the cameras fail to show up (lsusb shows none)
- I reboot, both cameras show up
- I launch the ROS nodelet manager for each camera with the device reset patch and this succeeded after 0-3 attempts (nodelet group is reset by an external watchdog I wrote if no output for 10 seconds).

This pattern repeats, almost every time (733/753 reboot cycles over the weekend). The remaining 20 boots show the cameras enumerating correctly on USB.

RealSense-Customer-Engineering, 2018-06-15:
[Realsense Customer Engineering Team Comment]
Hello **josephduchesne,**

**Please retest using the latest FW, v 5.9.11. You can download it here: https://downloadcenter.intel.com/download/27792/Latest-Firmware-for-Intel-RealSense-D400-Product-Family**

Regards,
Jesus G.
Intel Customer Support

sradmard, 2018-06-28:
Hi,
I am using the D435 on Jetson TX2 developer kit, using Ubuntu 16.04. My D435 is directly connected to the single USB 3 port of Jetson TX2. I have updated the camera FW to the FW, v 5.9.11 while still using the librealsense v2.10.4. However, the problem still persists for me, and unless I unplug and plug the camera back in, the camera fails to boot at every other run.

pumaking, 2018-07-14:
@sradmard I've had issues running the camera directly connected to the Jetson TX2 but none when connecting the camera to a powered USB 3.0 hub that is connected to the Jetson. I've also had the camera fail to stream on alternating runs when using Linux 4 Tegra 27. Upgrading to 28.2.1 solved this problem for me.

closed 2018-08-06 by RealSense-Customer-Engineering

josephduchesne, 2018-08-06:
I tried 5.9.14 without success. Without the USB hub it still fails to show up on boot a large percentage of the time.

wglockner, 2025-11-14:
Having the same issue.

MartyG-RealSense, 2025-11-15:
@wglockner  Have you tried resetting the camera automatically during your ROS launch by adding the parameter `initial_reset:=true` to your ROS launch command, please?