realsenseai/librealsense #1214: Any guidelines to integrate librealsense2 with eclipse in Ubuntu 16

VishalShelke, 2018-02-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

dorodnic, 2018-02-23:
Hi @VishalShelke 
We provide portable CMake scripts to let everyone use an IDE they prefer. As a tested and free option we recommend [Qt Creator](https://wiki.qt.io/Install_Qt_5_on_Ubuntu). 
You should however be able to configure Eclipse with CMake (you might find [this page](http://johnnado.com/use-cmake-with-eclipse/) useful). Let us know if you encounter any issues with that.

RealSense-Customer-Engineering, 2018-03-02:
[Realsense Customer Engineering Team Comment] 
@VishalShelke  Would like to know if you still need help with this.

closed 2018-03-19 by dorodnic