realsenseai/librealsense #1215: The rs-fw-logger's output looks like ......?

zenving, 2018-02-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415| 
| Firmware Version                   |05.08.15.00| 
| Operating System & Version |   Linux  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description

How to use rs-fw-logger Tool ?
Why I get the output below so ...?

Example

Waiting for RealSense device to connect...
RealSense device was connected...
Device Name: Intel RealSense 415
Device Location: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video1

2018-02-23 11:06:54  FW_Log_Data:0F 00 00 00 A0 02 05 10 50 00 83 F0 63 0A 00 00 00 00 00 00 9D C9 C1 3B A0 01 0A 10 40 02 2C 71 08 00 19 00 00 00 00 00 FA 52 B2 71 A0 01 0A 10 40 02 2C 81 04 00 08 00 00 00 00 00 39 02 DC 71 A0 01 23 10 23 00 6F 91 FF 57 00 00 00 00 00 00 8F 12 DC 71 A0 01 27 10 35 00 33 A3 00 00 4D 80 60 00 00 00 A7 14 DC 71 A0 01 26 10 E1 00 C8 B1 01 00 1D 03 AE 02 00 00 7C B7 5B 72 A0 01 3F 10 08 00 AC C0 1E 00 BE 08 00 00 00 00 F5 5A 5C 72 A0 01 0A 10 24 00 3D D1 01 00 00 00 42 00 00 00 27 00 5F 72 A0 01 0A 10 40 02 2C E1 08 00 20 00 00 00 00 00 37 F3 04 73 A0 01 0A 10 40 02 2C F1 04 00 08 00 00 00 00 00 B5 A1 2E 73 A0 01 0A 10 40 02 2C 01 08 00 19 00 00 00 00 00 4A 42 AE 73 A0 01 0A 10 40 02 2C 11 04 00 08 00 00 00 00 00 89 F1 D7 73 A0 01 23 10 23 00 6F 21 FF 57 00 00 00 00 00 00 DF 01 D8 73 A0 01 27 10 35 00 33 33 00 00 4D 80 30 00 00 00 F7 03 D8 73 A0 01 26 10 E1 00 C8 41 01 00 F7 00 D4 02 00 00 C0 A6 57 74 A0 01 3F 10 08 00 AC 50 1E 00 BE 08 00 00 00 00 D9 49 58 74 A0 01 0A 10 24 00 3D 61 01 00 00 00 42 00 00 00 65 F1 5A 74 A0 01 0A 10 40 02 2C 71 08 00 20 00 00 00 00 00 7C E2 00 75 A0 01 0A 10 40 02 2C 81 04 00 08 00 00 00 00 00 01 91 2A 75 A0 01 0A 10 40 02 2C 91 08 00 19 00 00 00 00 00 96 31 AA 75 A0 01 0A 10 40 02 2C A1 04 00 08 00 00 00 00 00 C1 E0 D3 75 A0 01 23 10 23 00 6F B1 FF 57 00 00 00 00 00 00 17 F1 D3 75 A0 01 27 10 35 00 33 C3 00 00 4D 80 4E 00 00 00 2F F3 D3 75 A0 01 26 10 E1 00 C8 D1 01 00 B6 02 A8 02 00 00 0A 96 53 76 A0 01 3F 10 08 00 AC E0 1E 00 BE 08 00 00 00 00 89 39 54 76 

2018-02-23 11:06:54  FW_Log_Data:0F 00 00 00 A0 01 0A 10 24 00 3D F1 01 00 00 00 42 00 00 00 0D DF 56 76 A0 01 0A 10 40 02 2C 01 08 00 20 00 00 00 00 00 CF D1 FC 76 A0 01 0A 10 40 02 2C 11 04 00 08 00 00 00 00 00 49 80 26 77 A0 01 0A 10 40 02 2C 21 08 00 19 00 00 00 00 00 E2 20 A6 77 A0 01 0A 10 40 02 2C 31 04 00 08 00 00 00 00 00 11 D0 CF 77 A0 01 23 10 23 00 6F 41 FF 57 00 00 00 00 00 00 67 E0 CF 77 A0 01 27 10 35 00 33 53 00 00 4D 80 2D 00 00 00 7F E2 CF 77 A0 01 26 10 E1 00 C8 61 01 00 6B 01 AF 02 00 00 BC 84 4F 78 A0 01 3F 10 08 00 AC 70 1E 00 BE 08 00 00 00 00 39 28 50 78 A0 01 0A 10 24 00 3D 81 01 00 00 00 42 00 00 00 FA CD 52 78 A0 01 0A 10 40 02 2C 91 08 00 20 00 00 00 00 00 1B C1 F8 78 A0 01 0A 10 40 02 2C A1 04 00 08 00 00 00 00 00 95 6F 22 79 A0 01 0A 10 40 02 2C B1 08 00 19 00 00 00 00 00 2E 10 A2 79 A0 01 0A 10 40 02 2C C1 04 00 08 00 00 00 00 00 61 BF CB 79 A0 01 23 10 23 00 6F D1 FF 57 00 00 00 00 00 00 C3 CF CB 79 A0 01 27 10 35 00 33 E3 00 00 4D 80 31 00 00 00 DB D1 CB 79 A0 01 26 10 E1 00 C8 F1 01 00 FF 00 D3 02 00 00 A1 74 4B 7A A0 01 3F 10 08 00 AC 00 1E 00 BE 08 00 00 00 00 2D 18 4C 7A A0 01 0A 10 24 00 3D 11 01 00 00 00 42 00 00 00 B1 BD 4E 7A A0 01 0A 10 40 02 2C 21 08 00 20 00 00 00 00 00 6B B0 F4 7A A0 01 0A 10 40 02 2C 31 04 00 08 00 00 00 00 00 E1 5E 1E 7B A0 01 0A 10 40 02 2C 41 08 00 19 00 00 00 00 00 76 FF 9D 7B A0 01 0A 10 40 02 2C 51 04 00 08 00 00 00 00 00 A5 AE C7 7B A0 01 23 10 23 00 6F 61 FF 57 00 00 00 00 00 00 FB BE C7 7B A0 01 27 10 35 00 33 73 00 00 4D 80 4D 00 00 00 13 C1 C7 7B 

2018-02-23 11:06:54  FW_Log_Data:0F 00 00 00 A0 01 26 10 E1 00 C8 81 01 00 62 01 DC 02 00 00 53 63 47 7C A0 01 3F 10 08 00 AC 90 1E 00 BE 08 00 00 00 00 DD 06 48 7C A0 01 0A 10 24 00 3D A1 01 00 00 00 42 00 00 00 0D AC 4A 7C A0 01 0A 10 40 02 2C B1 08 00 20 00 00 00 00 00 B3 9F F0 7C A0 01 0A 10 40 02 2C C1 04 00 08 00 00 00 00 00 31 4E 1A 7D A0 01 0A 10 40 02 2C D1 08 00 19 00 00 00 00 00 C6 EE 99 7D A0 01 0A 10 40 02 2C E1 04 00 08 00 00 00 00 00 F5 9D C3 7D A0 01 23 10 23 00 6F F1 FF 57 00 00 00 00 00 00 4B AE C3 7D A0 01 27 10 35 00 33 03 00 00 4D 80 56 00 00 00 63 B0 C3 7D A0 01 26 10 E1 00 C8 11 01 00 01 03 A8 02 00 00 3A 53 43 7E A0 01 3F 10 08 00 AC 20 1E 00 BE 08 00 00 00 00 B9 F6 43 7E A0 01 0A 10 24 00 3D 31 01 00 00 00 42 00 00 00 3D 9C 46 7E A0 01 0A 10 40 02 2C 41 08 00 20 00 00 00 00 00 03 8F EC 7E A0 01 0A 10 40 02 2C 51 04 00 08 00 00 00 00 00 79 3D 16 7F A0 01 0A 10 40 02 2C 61 08 00 19 00 00 00 00 00 0E DE 95 7F A0 01 0A 10 40 02 2C 71 04 00 08 00 00 00 00 00 41 8D BF 7F A0 01 23 10 23 00 6F 81 FF 57 00 00 00 00 00 00 97 9D BF 7F A0 01 27 10 35 00 33 93 00 00 4D 80 2D 00 00 00 AF 9F BF 7F A0 01 26 10 E1 00 C8 A1 01 00 92 05 42 02 00 00 86 42 3F 80 A0 01 3F 10 08 00 AC B0 1E 00 BE 08 00 00 00 00 0D E6 3F 80 A0 01 0A 10 24 00 3D C1 01 00 00 00 42 00 00 00 FF 8A 42 80 A0 01 0A 10 40 02 2C D1 08 00 20 00 00 00 00 00 4B 7E E8 80 A0 01 0A 10 40 02 2C E1 04 00 08 00 00 00 00 00 C5 2C 12 81 A0 01 0A 10 40 02 2C F1 08 00 19 00 00 00 00 00 5E CD 91 81 A0 01 0A 10 40 02 2C 01 04 00 08 00 00 00 00 00 8D 7C BB 81 

2018-02-23 11:06:54  FW_Log_Data:0F 00 00 00 A0 01 23 10 23 00 6F 11 FF 57 00 00 00 00 00 00 E3 8C BB 81 A0 01 27 10 35 00 33 23 00 00 98 66 10 00 00 00 FB 8E BB 81 A0 01 26 10 E1 00 C8 31 01 00 3A 04 F2 01 00 00 D4 31 3B 82 A0 01 3F 10 08 00 AC 40 1E 00 BE 08 00 00 00 00 4D D5 3B 82 A0 01 0A 10 24 00 3D 51 01 00 00 00 42 00 00 00 D1 7A 3E 82 A0 01 0A 10 40 02 2C 61 08 00 20 00 00 00 00 00 9B 6D E4 82 A0 01 0A 10 40 02 2C 71 04 00 08 00 00 00 00 00 15 1C 0E 83 A0 01 0A 10 40 02 2C 81 08 00 19 00 00 00 00 00 AA BC 8D 83 A0 01 0A 10 40 02 2C 91 04 00 08 00 00 00 00 00 D9 6B B7 83 A0 01 23 10 23 00 6F A1 FF 57 00 00 00 00 00 00 2F 7C B7 83 A0 01 27 10 35 00 33 B3 00 00 FA 25 10 00 00 00 47 7E B7 83 A0 01 26 10 E1 00 C8 C1 01 00 AF 00 34 02 00 00 15 21 37 84 A0 01 3F 10 08 00 AC D0 1E 00 BE 08 00 00 00 00 9D C4 37 84 A0 01 0A 10 24 00 3D E1 01 00 00 00 42 00 00 00 11 6A 3A 84 A0 01 0A 10 40 02 2C F1 08 00 20 00 00 00 00 00 E7 5C E0 84 A0 01 0A 10 40 02 2C 01 04 00 08 00 00 00 00 00 61 0B 0A 85 A0 01 0A 10 40 02 2C 11 08 00 19 00 00 00 00 00 F6 AB 89 85 A0 01 0A 10 40 02 2C 21 04 00 08 00 00 00 00 00 25 5B B3 85 A0 01 23 10 23 00 6F 31 FF 57 00 00 00 00 00 00 7B 6B B3 85 A0 01 27 10 35 00 33 43 00 00 37 56 10 00 00 00 93 6D B3 85 A0 01 26 10 E1 00 C8 51 01 00 94 00 84 02 00 00 61 10 33 86 A0 01 3F 10 08 00 AC 60 1E 00 BE 08 00 00 00 00 D9 B3 33 86 A0 01 0A 10 24 00 3D 71 01 00 00 00 42 00 00 00 D6 58 36 86 A0 01 0A 10 40 02 2C 81 08 00 20 00 00 00 00 00 33 4C DC 86 A0 01 0A 10 40 02 2C 91 04 00 08 00 00 00 00 00 BD FA 05 87 

2018-02-23 11:06:54  FW_Log_Data:0F 00 00 00 A0 01 0A 10 40 02 2C A1 08 00 19 00 00 00 00 00 3E 9B 85 87 A0 01 0A 10 40 02 2C B1 04 00 08 00 00 00 00 00 71 4A AF 87 A0 01 23 10 23 00 6F C1 FF 57 00 00 00 00 00 00 C7 5A AF 87 A0 01 27 10 35 00 33 D3 00 00 4D 80 1D 00 00 00 DF 5C AF 87 A0 01 26 10 E1 00 C8 E1 01 00 D4 00 B7 02 00 00 B9 FF 2E 88 A0 01 3F 10 08 00 AC F0 1E 00 BE 08 00 00 00 00 B9 A5 2F 88 A0 01 0A 10 24 00 3D 01 01 00 00 00 42 00 00 00 3D 4B 32 88 A0 01 0A 10 40 02 2C 11 08 00 20 00 00 00 00 00 7F 3B D8 88 A0 01 0A 10 40 02 2C 21 04 00 08 00 00 00 00 00 05 EA 01 89 A0 01 0A 10 40 02 2C 31 08 00 19 00 00 00 00 00 8A 8A 81 89 A0 01 0A 10 40 02 2C 41 04 00 08 00 00 00 00 00 BD 39 AB 89 A0 01 23 10 23 00 6F 51 FF 57 00 00 00 00 00 00 13 4A AB 89 A0 01 27 10 35 00 33 63 00 00 4D 80 36 00 00 00 2B 4C AB 89 A0 01 26 10 E1 00 C8 71 01 00 1C 01 D2 02 00 00 EA EE 2A 8A A0 01 3F 10 08 00 AC 80 1E 00 BE 08 00 00 00 00 CD 92 2B 8A A0 01 0A 10 24 00 3D 91 01 00 00 00 42 00 00 00 BE 37 2E 8A A0 01 0A 10 40 02 2C A1 08 00 20 00 00 00 00 00 C7 2A D4 8A A0 01 0A 10 40 02 2C B1 04 00 08 00 00 00 00 00 55 D9 FD 8A A0 01 0A 10 40 02 2C C1 08 00 19 00 00 00 00 00 DA 79 7D 8B A0 01 0A 10 40 02 2C D1 04 00 08 00 00 00 00 00 09 29 A7 8B A0 01 23 10 23 00 6F E1 FF 57 00 00 00 00 00 00 5F 39 A7 8B A0 01 27 10 35 00 33 F3 00 00 4D 80 4C 00 00 00 77 3B A7 8B A0 01 26 10 E1 00 C8 01 01 00 5D 01 DC 02 00 00 B4 DD 26 8C A0 01 3F 10 08 00 AC 10 1E 00 BE 08 00 00 00 00 3D 81 27 8C A0 01 0A 10 24 00 3D 21 01 00 00 00 42 00 00 00 6D 26 2A 8C

dorodnic, 2018-02-23:
Hi @zenving 

The goal of firmware logger tool is to let you, the end user, collect troubleshooting logs and submit them back to Intel, in case of a serious bug. Internally, we have tools to decode these into something firmware engineers can understand. 

Of course no private information is being collected and the data is not encrypted in any way. All the key structs and enumerations are public inside the source code. What changes from firmware to firmware are the exact formatting strings for each output.

zenving, 2018-02-23:
Hi @dorodnic
  Thanks for your reply, but do you mean that the tool's output is only can be understand by the Intel's engineers ? In other word, it's not useful for user generally, but if I want get the camera's some useful log information like fps every frame's coming time etc., what should I do.

dorodnic, 2018-02-25:
Yes, you are correct. We discussed publishing firmware log structure but came to the conclusion that it would not offer anything of value to an end-user. 
Please check out [enabling uvcvideo traces](https://github.com/IntelRealSense/librealsense/blob/master/doc/troubleshooting.md#uvc-video-module-traces) on Linux, it might be what you need.

closed 2018-02-26 by dorodnic