realsenseai/librealsense #1216: CMakeLists.txt: cross compilation for raspberry with yocto fails

giongieri, 2018-02-23:
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Ubuntu 16.04 LTS)  | 
| Kernel Version (Linux Only)    |  (4.4.0-112)                                         | 
| Platform                                 | PC/Raspberry Pi  |
| SDK Version                            |  }                          | 

### Issue Description
cross compilation for raspberry pi 3 with yocto layer meta-intel-realsense fails because of CMakeLists.txt line 590 ff. (librealsense v2.10.0):

 else(${MACHINE} MATCHES "arm-linux-gnueabihf")
      set(CMAKE_C_FLAGS   "${CMAKE_C_FLAGS}   -mssse3")
      set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mssse3")
endif(${MACHINE} MATCHES "arm-linux-gnueabihf")

-mssse3 Flag doesnt work for cross compilation, else statement is strange...
Works with following:

    if(${MACHINE} MATCHES "arm-linux-gnueabihf")
      set(CMAKE_C_FLAGS   "${CMAKE_C_FLAGS}   -mfpu=neon -mfloat-abi=hard -ftree-vectorize")
      set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mfpu=neon -mfloat-abi=hard -ftree-vectorize")
    elseif(${MACHINE} MATCHES "aarch64-linux-gnu")
      set(CMAKE_C_FLAGS   "${CMAKE_C_FLAGS}   -mstrict-align -ftree-vectorize")
      set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mstrict-align -ftree-vectorize")
    else()
      set(CMAKE_C_FLAGS   "${CMAKE_C_FLAGS}   -mssse3")
      set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mssse3")
endif(${MACHINE} MATCHES "arm-linux-gnueabihf")

Build directly on raspberry pi works (with mate Desktop Environment 1.16.2), but it seems that cmake ignores wrong else-statement.

giongieri, 2018-02-23:
found problem: machine is  arm-poky-linux-gnueabi, so this does fit else statement. Directly on raspberry pi, machine is arm-linux-gnueabihf, this fits if statement.

anyway, it makes no sense to do it this way:

    if(${MACHINE} MATCHES "arm-linux-gnueabihf")
      set(CMAKE_C_FLAGS   "${CMAKE_C_FLAGS}   -mfpu=neon -mfloat-abi=hard -ftree-vectorize")
      set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mfpu=neon -mfloat-abi=hard -ftree-vectorize")
    elseif(${MACHINE} MATCHES "aarch64-linux-gnu")
      set(CMAKE_C_FLAGS   "${CMAKE_C_FLAGS}   -mstrict-align -ftree-vectorize")
      set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mstrict-align -ftree-vectorize")
    else(${MACHINE} MATCHES "arm-linux-gnueabihf")
      set(CMAKE_C_FLAGS   "${CMAKE_C_FLAGS}   -mssse3")
      set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -mssse3")
    endif(${MACHINE} MATCHES "arm-linux-gnueabihf")

i suggest: 
either put -mssse3 flags into if statement or leave content in () of else statement.

dorodnic, 2018-02-23:
Hi @giongieri 
Glad you were able to advance. We are trying to integrate RealSense with as many platforms as possible, but can't test all possible combinations. You are welcomed to submit PR with changes to `CMakeLists.txt` and documentation of your findings.

giongieri, 2018-02-23:
ok.

closed 2018-02-27 by jlgarcia75