realsenseai/librealsense #1217: Using advanced mode presets via C API

SirDifferential, 2018-02-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D415 / D435 } | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Win 8.1 / Debian 9 | 
| Kernel Version (Linux Only)    | 4.9.65 with the patches applied | 
| Platform                                 | PC |
| SDK Version                            | 2.0 on Windows, latest master on Linux | 

### Issue Description

I'm trying to use the advanced mode presets through the C API, but I don't see any changes in my depth feed. Using the realsense viewer tool I can see the presets are being utilized correctly. Here's my somewhat minimal test program I wrote to showcase this problem: https://github.com/SirDifferential/minimal_realsense2/blob/master/main.c#L610

Compile with: `make` or gcc -I/path/to/librealsense/include -L/path/to/librealsense/build -lrealsense2 -lSDL2
Run with: LD_LIBRARY_PATH=/path/to/librealsense/build ./a.out (exit with ^C)

I can manually set the assorted advanced mode parameters and toggle the IR emitter using the `set_option` calls.

dorodnic, 2018-02-23:
Hi @SirDifferential 
I suspect that with our current implementation, you should issue `set_option` on the same `sensor` object used for streaming, otherwise it will not apply. 
When you select a certain preset (say "High Accuracy") we can't yet issue the command to the firmware because different settings will need to be selected depending on the resolution. So we store the value inside the `sensor` object and apply it once you start streaming. If `sensor` gets deleted, no action is taken.

SirDifferential, 2018-02-23:
Something like this should work?

```
rs2_sensor_list* sensor_list = rs2_query_sensors(dev, NULL);
int sensor_list_count = rs2_get_sensors_count(sensor_list, NULL);

const char* new_preset = "High accuracy";
int8_t done = 0;

for (int sensor = 0; sensor < sensor_list_count; sensor++)
{
    if (done == 1)
        break;

    rs2_sensor* sen = rs2_create_sensor(sensor_list, sensor, &e);
    float min, max, step, def;
    rs2_get_option_range((const rs2_options*)sen, RS2_OPTION_VISUAL_PRESET, &min, &max, &step, &def, NULL);
    for (r = (int)min; r < (int)max; r++) {
        if (strcmp(preset_desc, new_preset) == 0) {
            rs2_set_option((const rs2_options*)sen, RS2_OPTION_VISUAL_PRESET, preset_index, NULL);
            done = 1;
            break;
        }
    }

    // store sen, delete on program exit
}

rs2_pipeline* pipe = rs2_create_pipeline(ctx, &e);
rs2_config* config = rs2_create_config(&e);
rs2_config_enable_stream(config, RS2_STREAM_DEPTH, -1, cDepthW, cDepthH, RS2_FORMAT_Z16, 30, NULL);
rs2_pipeline_profile* selection = rs2_pipeline_start_with_config(pipe, config, NULL);
```

I've changed the minimal program to essentially do this - query the device for `rs2_sensor` instances, set their preset, then start the pipeline. Every time the preset needs to be changed, I destroy the entire state and start from scratch doing the whole setup again. I still don't see the presets changing, unfortunately.

dorodnic, 2018-02-24:
Please give this a try:
```c
rs2_pipeline* pipe = rs2_create_pipeline(ctx, NULL);
rs2_config* config = rs2_create_config(NULL);
rs2_config_enable_stream(config, RS2_STREAM_DEPTH, -1, cDepthW, cDepthH, RS2_FORMAT_Z16, 30, NULL);

rs2_pipeline_profile* selection = rs2_config_resolve(config, pipe, NULL);
rs2_device* dev = rs2_pipeline_profile_get_device(config, NULL);

rs2_sensor_list* sensor_list = rs2_query_sensors(dev, NULL);
int sensor_list_count = rs2_get_sensors_count(sensor_list, NULL);

const char* new_preset = "High accuracy";
int8_t done = 0;

for (int sensor = 0; sensor < sensor_list_count; sensor++)
{
    if (done == 1)
        break;

    rs2_sensor* sen = rs2_create_sensor(sensor_list, sensor, NULL);
    float min, max, step, def;
    rs2_get_option_range((const rs2_options*)sen, RS2_OPTION_VISUAL_PRESET, &min, &max, &step, &def, NULL);
    for (r = (int)min; r < (int)max; r++) {
        if (strcmp(preset_desc, new_preset) == 0) {
            rs2_set_option((const rs2_options*)sen, RS2_OPTION_VISUAL_PRESET, r, NULL);
            done = 1;
            break;
        }
    }
}

rs2_pipeline_start_with_config(pipe, config, NULL);
```

SirDifferential, 2018-02-26:
This did it. I can now see the presets changing successfully. Updated my minimal example in case someone else needs to get this working.

Thanks!

closed 2018-02-26 by SirDifferential

dorodnic, 2018-02-28:
We would love to link-back to [minimal_realsense2](https://github.com/SirDifferential/minimal_realsense2) from our C examples page to offer customers more options for getting started. Would that be Ok?

SirDifferential, 2018-03-01:
Yes, that's fine.