realsenseai/librealsense #1219: 8 RS400 cameras on one Intel NUC

maciekd, 2018-02-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 |
| Firmware Version                   | 05.08.09.00 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.4.0-116-generic | 
| Platform                                 | Intel NUC  |
| SDK Version                            | 2.10.0 }                          | 

We've been successfully running 7 R200 cameras on one NUC for around a year. We plan on migrating our system to the new RS400 cameras.

We use rather low framerates and resolution on most cameras so bandwidth is not an issue. We use powered hubs in order to power all cameras sufficiently. Once we fixed all the kinks this setup worked very well.

We plan on running the same setup but with RS400 cameras (a mix of RS415 and RS435). We know that the new cameras use more power but since our hubs are externally powered we hope everything should work well.

Just wanted to check this setup seems plausible with the new cameras?

dorodnic, 2018-02-23:
Hi @maciekd 
Our CTO group has previously published results of running 4 cameras:
[LinkedIn link](https://www.linkedin.com/feed/update/urn:li:activity:6360930349748486144)
I can confirm that there are no hard software limitations to it, however, working with multiple devices involves many practical engineering challenges.
We will do our best to assist.

mhkabir, 2018-02-23:
@maciekd Would you be able to share your hub setup and a bit more about how you connect your cameras to the NUC? Thanks!

maciekd, 2018-02-26:
Out setup was for the R200 cameras. We used two four-port USB hubs. Not powered. Tree cameras in each hub and one camera directly in the NUC. The R200 cameras use about 0.3A, so we used 0.9A per hub/NUC port (which is exactly the spec limit for USB 3).

More than half of the 10.000 autonomous flights we did in 2017 where using this setup. Might be hard to see computer details but here is some video: https://www.youtube.com/watch?v=jmZi3Jclo-w

The RS400 series use around twice the power (to the best of our knowledge) so powered hubs will probably be a must.

mhkabir, 2018-02-27:
@maciekd Thanks! Looks great!
Off topic, but I'm curious - how do you deal with the GPS interference from the R200s USB3 bus?

maciekd, 2018-02-27:
GPS interference is a big issue. There's no foolproof answer but generally:
1) Move the GPS as far away as possible. 
2) Shield the USB connectors, USB hubs, USB cables and cameras (in that order).

maciekd, 2018-03-02:
Just wanted to let you guys now that we're now running 7 RS400 cameras on an Intel NUC successfully. We still have issue #1213 but other that that it seems to be working very nicely. Superb coverage.

Here's a video of our UI running with the new cameras (almost 100% coverage!):
https://www.dropbox.com/s/kavbyhxwkxy7hlj/depth%202.0.mov?dl=0

agrunnet, 2018-03-02:
Very cool. 
One observations: It does look you are plotting them as depth-maps in 3D rather than calculating the point-clouds and then rotating them and adding them to 3D space.  You should not see the hexagonal pattern in the far walls or issues with overlaps if you first convert each camera to a 3D point cloud and add them.  Also, I presume this was low resolution of VGA or lower? You will be best depth with higher resolution. Of course that becomes an issue for the USB bandwidth. The RealSense team is working on adding HW sync and compression that will allow more cameras to operate at full resolution.

maciekd, 2018-03-05:
Super, thanks for the tips! Will have a look at this. 

We're running at low resolution, yes. In our application we have no use for higher res.

closed 2018-03-19 by dorodnic