realsenseai/librealsense #1220: No UVC backend selected.

pedrogasg, 2018-02-23:
```Scanning dependencies of target pyrealsense2
[  5%] Building CXX object CMakeFiles/pyrealsense2.dir/python.cpp.o
[ 11%] Linking CXX shared library pyrealsense2.cpython-36m-x86_64-linux-gnu.so
[ 11%] Built target pyrealsense2
Scanning dependencies of target pybackend2
[ 17%] Building CXX object CMakeFiles/pybackend2.dir/home/real/librealsense/src/backend.cpp.o
/home/real/librealsense/src/backend.cpp:11:2: error: #error No UVC backend selected. Please #define exactly one of RS2_USE_LIBUVC_BACKEND, RS2_USE_WMF_BACKEND, or RS2_USE_V4L2_BACKEND
 #error No UVC backend selected. Please #define exactly one of RS2_USE_LIBUVC_BA
  ^
CMakeFiles/pybackend2.dir/build.make:62: recipe for target 'CMakeFiles/pybackend2.dir/home/real/librealsense/src/backend.cpp.o' failed
make[2]: *** [CMakeFiles/pybackend2.dir/home/real/librealsense/src/backend.cpp.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/pybackend2.dir/all' failed
make[1]: *** [CMakeFiles/pybackend2.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |    D400   | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |  Linux (Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.10.0-28-generic                                      | 
| Platform                                 | PC|
| SDK Version                            |  git                         | 

### Issue Description
<Describe your issue / question / feature request / etc..>

zivsha, 2018-02-24:
Are you building the main `Cmakelists.txt` or the one inside the python folder? 
You should build the main one and only pass the `-DBUILD_PYTHON_BINDINGS=ON` flag.

pedrogasg, 2018-02-24:
`or the one inside the python folder` <- Yeah this one was my problem, thanks for the help 
It works like a charm once you add the
 `-DBUILD_PYTHON_BINDINGS=bool:true` 
and the version on python you are using 
`-DPYTHON_EXECUTABLE=/home/real/miniconda3/envs/py36/bin/python` 
miniconda python 3.6 in my case

One question though is necesary to add all the suffix to the build ?

My build name -> `pyrealsense2.cpython-36m-x86_64-linux-gnu.so`

Thanks again

closed 2018-02-26 by jlgarcia75