realsenseai/librealsense #1223: Some question about rs2::colorizer ?

zenving, 2018-02-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
In the wrappers of opencv, there are one example named GrabCut
I have a question about the rs2::colorizer 's set_option ...
like this 

   `rs2::colorizer colorize;`

  `  colorize.set_option (RS2_OPTION_COLOR_SCHEME, 2);`
The question is that what's the numeber 2 mean, and the function` rs2::colorizer` 's real work way, in other word, how does it work ? In my test, I think it's alway do histogram equalization, right , but how can I change the output color scheme ? I will appreciate if there are some one can help me .

dorodnic, 2018-02-24:
Hi @zenving 

First, regarding the functionality of the `colorizer`:
1. By default, it will convert any 16-bit depth image to an RGB image using the JET color scheme. 
2. You can enable / disable histogram equalization using `RS2_OPTION_HISTOGRAM_EQUALIZATION_ENABLED` option. Generally speaking, histogram equalization produce images with more details, but it can also exaggerate some of the noise if the dynamic range is low. 
3. When histogram equalization is disabled, the color map will be scaled from `RS2_OPTION_MIN_DISTANCE` to `RS2_OPTION_MAX_DISTANCE` in meters.
4. We implemented several color maps you can use out of the box by settings the `RS2_OPTION_COLOR_SCHEME` option. You can get the total number of color maps using `get_option_range` method.

Now regarding the use of the `colorizer` in code: We often use `colorizer` to shape depth data into something the user can more easily process. For consistency and clarity we use the `colorizer` even when better methods exist (like with the Unity wrapper, where it makes much more sense to implement this conversion in a shader, or in OpenCV examples, where you can achieve a similar effect with couple of OpenCV calls). 

I somewhat regret using the `colorizer` in the Grab-Cut example. Sure, it works, and we saw an opportunity to showcase RealSense API, but in context of OpenCV example it would make more sense to use standard OpenCV tools.

closed 2018-02-25 by zenving

bob-robster, 2019-03-09:
Can the colorizer be used on _Y8 frames as well? I would like to add pseudocolor to the IR images (which are monochrome on the D435).