realsenseai/librealsense #1225: installation qestion ------ could not insert 'videodev'

GITSHOHOKU, 2018-02-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300| 
| Firmware Version                   | 2.10.0 | 
| Operating System & Version |   Ubuntu 14.04  | 
| Kernel Version (Linux Only)    |  4.4.0-116-generic                                       | 
| Platform                                 | Intel NUC7i7BNH  |
| SDK Version                            | 2.0                    | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I had installed the SDK on Intel NUC ,just only followed steps here  https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md .The error never showed when I installed on my own PC. When I installed it at the last step ./scripts/patch-realsense-ubuntu-xenial.sh the error showed as below:
Patched kernels modules were created successfully

Handle videodev :
Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Exec format error
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  videodev  module was reloaded

Anybody know why? Please help me with this problem.Did it because the platform change?

ev-mp, 2018-02-25:
Hello @GITSHOHOKU , the error may indicate that the patched module kernel was build for a version different from the currently loaded kernel. Make sure to follow the instructions and [update Ubuntu](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#make-ubuntu-up-to-date) before applying the kernel patch.  

Additionally, please provide the following data:
1. `uname -r` 
2. `dmesg` output after the patch script completes.

GITSHOHOKU, 2018-02-26:
I tried it again and I am sure I follow the step to update ubuntu ,
1. when I ran uname -r I got 
Patched kernels modules were created successfully

Handle videodev :
Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Exec format error
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
The original  videodev  module was reloaded
hkwal@MyUbuntu:~/librealsense-master$ uname -r
4.4.0-116-generic

2.last 50 lines of dmesg :
```
hkwal@MyUbuntu:~/librealsense-master$ sudo dmesg | tail -n 50
[    4.701009] snd_hda_codec_realtek hdaudioC0D0:    hp_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    4.701010] snd_hda_codec_realtek hdaudioC0D0:    mono: mono_out=0x0
[    4.701011] snd_hda_codec_realtek hdaudioC0D0:    inputs:
[    4.701012] snd_hda_codec_realtek hdaudioC0D0:      Mic=0x12
[    4.749834] input: HDA Intel PCH Front Headphone as /devices/pci0000:00/0000:00:1f.3/sound/card0/input8
[    4.749874] input: HDA Intel PCH HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input9
[    4.749907] input: HDA Intel PCH HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input10
[    4.749944] input: HDA Intel PCH HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input11
[    4.763322] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    4.794715] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[    4.794857] usbcore: registered new interface driver mt7601u
[    4.821469] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    5.238206] init: plymouth-upstart-bridge main process ended, respawning
[    5.242698] init: plymouth-upstart-bridge main process (1161) terminated with status 1
[    5.242707] init: plymouth-upstart-bridge main process ended, respawning
[    5.300095] scsi 1:0:0:0: Direct-Access     Xiaomi   WiFi Flash Disk  1.00 PQ: 0 ANSI: 2
[    5.301926] sd 1:0:0:0: Attached scsi generic sg0 type 0
[    5.303445] sd 1:0:0:0: [sda] 15669248 512-byte logical blocks: (8.02 GB/7.47 GiB)
[    5.303697] sd 1:0:0:0: [sda] Write Protect is off
[    5.303700] sd 1:0:0:0: [sda] Mode Sense: 03 00 00 00
[    5.303927] sd 1:0:0:0: [sda] No Caching mode page found
[    5.303930] sd 1:0:0:0: [sda] Assuming drive cache: write through
[    5.306945]  sda: sda4
[    5.309110] sd 1:0:0:0: [sda] Attached SCSI removable disk
[    5.753056] wlan0: authenticate with 50:fa:84:64:ab:be
[    5.764889] wlan0: send auth to 50:fa:84:64:ab:be (try 1/3)
[    5.766352] wlan0: authenticated
[    5.767184] wlan0: associate with 50:fa:84:64:ab:be (try 1/3)
[    5.770897] wlan0: RX AssocResp from 50:fa:84:64:ab:be (capab=0x411 status=0 aid=1)
[    5.805347] wlan0: associated
[    5.805377] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[    5.807281] cfg80211: Regulatory domain changed to country: CN
[    5.807284] cfg80211:  DFS Master region: unset
[    5.807285] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[    5.807287] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    5.807288] cfg80211:   (5735000 KHz - 5835000 KHz @ 40000 KHz), (N/A, 3000 mBm), (N/A)
[    5.807290] cfg80211:   (57240000 KHz - 59400000 KHz @ 2160000 KHz), (N/A, 2800 mBm), (N/A)
[    5.807291] cfg80211:   (59400000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 4400 mBm), (N/A)
[    5.807292] cfg80211:   (63720000 KHz - 65880000 KHz @ 2160000 KHz), (N/A, 2800 mBm), (N/A)
[    5.815529] [drm] RC6 on
[   11.455175] FAT-fs (sda4): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   33.174802] systemd-hostnamed[2425]: Warning: nss-myhostname is not installed. Changing the local hostname might make it unresolveable. Please install nss-myhostname!
[   34.416835] audit_printk_skb: 51 callbacks suppressed
[   34.416838] audit: type=1400 audit(1519615334.216:28): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/lib/cups/backend/cups-pdf" pid=2501 comm="apparmor_parser"
[   34.416844] audit: type=1400 audit(1519615334.216:29): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/sbin/cupsd" pid=2501 comm="apparmor_parser"
[  368.612014] systemd-hostnamed[4295]: Warning: nss-myhostname is not installed. Changing the local hostname might make it unresolveable. Please install nss-myhostname!
[ 7551.145445] media: Linux media interface: v0.10
[ 7551.145565] _`videodev: version magic '4.4.0-116-generic SMP mod_unload modversions ' should be '4.4.0-116-generic SMP mod_unload modversions retpoline '_`
[ 7551.163588] Linux video capture interface: v2.00
[ 7881.233866] systemd-hostnamed[24433]: Warning: nss-myhostname is not installed. Changing the local hostname might make it unresolveable. Please install nss-myhostname!

```

GITSHOHOKU, 2018-02-26:
the problem seems to be indicated by this line 
_`videodev: version magic '4.4.0-116-generic SMP mod_unload modversions ' should be '4.4.0-116-generic SMP mod_unload modversions retpoline '_`
but I don't know where is wrong because I do update my Ubuntu

ev-mp, 2018-02-26:
Hello @GITSHOHOKU , I was able to reproduce the error with two 4.4.0-116-kernel based desktops, but then I found [this link](https://www.systutorials.com/linux-kernels/55769/revert-module-add-retpoline-tag-to-vermagic-linux-4-9-79/) that indicates that some previous decision in Linux tree was reverted and a patch has been removed from the kernel tree (see [this](https://patchwork.kernel.org/patch/10190481/) for kernel 4.4).
Apparently the change in vermagic string - `4.4.0-116-generic SMP mod_unload modversions` instead of `4.4.0-116-generic SMP mod_unload modversions retpoline`  (Note the last word removed)
did not propagate to Ubuntu distro yet, but did reach the source code tree. Hence the generated error.

I will work to establish a repro environment and update the thread when the fix is set.

In the meanwhile the alternatives are:
1 - Use librealsense [DKMS package](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) to install kernel patches:
>  sudo apt-get install realsense-uvcvideo
2. Switch to another supported LTS kernel, 4.10 or 4.13, and reapply the patch.

GITSHOHOKU, 2018-02-27:
Thanks for your reply ev-mp ,I ues `sudo apt-get install realsense-uvcvideo`to install kernel patches
but I got error again 
uvcvideo:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/4.4.0-116-generic/updates/dkms/

depmod....

DKMS: install completed.
Loading new realsense-uvcvideo-4.10.0 DKMS files...
First Installation: checking all kernels...
Building only for 4.4.0-116-generic
Building initial module for 4.4.0-116-generic
Error! Application of patch 10-realsense-camera-formats-4.4.patch failed.
Check /var/lib/dkms/realsense-uvcvideo/4.10.0/build/ for more information.
正在处理用于 shim-signed (1.32~14.04.2+0.9+1474479173.6c180c6-1ubuntu1) 的触发器 ...
Secure Boot not enabled on this system.
Did it mean I must change to another supported LTS kernel

ev-mp, 2018-02-27:
Hello @GITSHOHOKU , the warning you receive is a known issues as DKMS tries to install both for 4.4 and 4.10 kernels, (and fails for 4.10, which is not current). We are currently testing an updated DKMS that will remove this inconvenience. But the DKMS for 4.4 would be installed by then.
To verify that it is installed properly please proceed according to [DKMS installation check-list](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages):
>Complete the installation by performing
reboot
to reload the kernel with modules provided by DKMS.
Verify that the kernel is updated :
modinfo uvcvideo | grep "version:" should include realsense string

Then run `realsense-viewer` tool to stream depth/infrared and update

GITSHOHOKU, 2018-03-01:
Thanks for your `help` ev-mp. I didn't try after 
``modinfo uvcvideo | grep "version:" should include realsense string``  shows 4.4.0 
Then I changed my kernels to 4.10.0-28 (Ubuntu16.04) and follow the step when I patched module it shows
`Patched kernels modules were created successfully

Handle videodev :
Applying the patched module ...  succeeded

Handle uvcvideo :
Applying the patched module ...  succeeded

Handle hid-sensor-accel-3d :
Applying the patched module ...  succeeded

Handle hid-sensor-gyro-3d :
Applying the patched module ...  succeeded

Script has completed. Please consult the installation guide for further instruction.
`
Then I do dmsg
`[    2.511604] input: HDA Intel PCH Front Headphone as /devices/pci0000:00/0000:00:1f.3/sound/card0/input7
[    2.511647] input: HDA Intel PCH HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input8
[    2.511687] input: HDA Intel PCH HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input9
[    2.511726] input: HDA Intel PCH HDMI/DP,pcm=8 as /devices/pci0000:00/0000:00:1f.3/sound/card0/input10
[    2.620008] usb 3-1: Device not responding to setup address.
[    2.640521] audit: type=1400 audit(1519867363.321:2): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/sbin/dhclient" pid=677 comm="apparmor_parser"
[    2.640523] audit: type=1400 audit(1519867363.321:3): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/lib/NetworkManager/nm-dhcp-client.action" pid=677 comm="apparmor_parser"
[    2.640524] audit: type=1400 audit(1519867363.321:4): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/lib/NetworkManager/nm-dhcp-helper" pid=677 comm="apparmor_parser"
[    2.640524] audit: type=1400 audit(1519867363.321:5): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/lib/connman/scripts/dhclient-script" pid=677 comm="apparmor_parser"
[    2.640818] audit: type=1400 audit(1519867363.321:6): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/lib/lightdm/lightdm-guest-session" pid=676 comm="apparmor_parser"
[    2.640819] audit: type=1400 audit(1519867363.321:7): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/lib/lightdm/lightdm-guest-session//chromium" pid=676 comm="apparmor_parser"
[    2.641599] audit: type=1400 audit(1519867363.321:8): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/lib/snapd/snap-confine" pid=681 comm="apparmor_parser"
[    2.641600] audit: type=1400 audit(1519867363.321:9): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/lib/snapd/snap-confine//mount-namespace-capture-helper" pid=681 comm="apparmor_parser"
[    2.641890] audit: type=1400 audit(1519867363.321:10): apparmor="STATUS" operation="profile_load" profile="unconfined" name="webbrowser-app" pid=680 comm="apparmor_parser"
[    2.679116] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    2.679117] Bluetooth: BNEP filters: protocol multicast
[    2.679120] Bluetooth: BNEP socket layer initialized
[    2.757407] IPv6: ADDRCONF(NETDEV_UP): wlp58s0: link is not ready
[    2.759459] iwlwifi 0000:3a:00.0: L1 Disabled - LTR Enabled
[    2.759711] iwlwifi 0000:3a:00.0: L1 Disabled - LTR Enabled
[    2.816417] [drm] RC6 on
[    2.831958] usb 3-1: device not accepting address 4, error -71
[    2.875424] iwlwifi 0000:3a:00.0: L1 Disabled - LTR Enabled
[    2.875676] iwlwifi 0000:3a:00.0: L1 Disabled - LTR Enabled
[    2.929175] IPv6: ADDRCONF(NETDEV_UP): wlp58s0: link is not ready
[    2.931808] IPv6: ADDRCONF(NETDEV_UP): eno1: link is not ready
[    2.951953] usb 3-1: new full-speed USB device number 5 using xhci_hcd
[    2.952194] usb 3-1: Device not responding to setup address.
[    3.116057] IPv6: ADDRCONF(NETDEV_UP): eno1: link is not ready
[    3.159973] usb 3-1: Device not responding to setup address.
[    3.166258] IPv6: ADDRCONF(NETDEV_UP): wlp58s0: link is not ready
[    3.367954] usb 3-1: device not accepting address 5, error -71
[    3.367992] usb usb3-port1: unable to enumerate USB device
[    3.790222] Bluetooth: RFCOMM TTY layer initialized
[    3.790226] Bluetooth: RFCOMM socket layer initialized
[    3.790229] Bluetooth: RFCOMM ver 1.11
[   17.199985] wlp58s0: authenticate with b0:d5:9d:a7:c6:67
[   17.206376] wlp58s0: send auth to b0:d5:9d:a7:c6:67 (try 1/3)
[   17.238266] wlp58s0: authenticated
[   17.239849] wlp58s0: associate with b0:d5:9d:a7:c6:67 (try 1/3)
[   17.264824] wlp58s0: RX AssocResp from b0:d5:9d:a7:c6:67 (capab=0x431 status=0 aid=4)
[   17.264825] wlp58s0: invalid AID value 0x4; bits 15:14 not set
[   17.266514] wlp58s0: associated
[   17.266536] IPv6: ADDRCONF(NETDEV_CHANGE): wlp58s0: link becomes ready
[ 4291.673859] media: Linux media interface: v0.10
[ 4291.673939] videodev: loading out-of-tree module taints kernel.
[ 4291.673986] videodev: module verification failed: signature and/or required key missing - tainting kernel
[ 4291.675875] Linux video capture interface: v2.00
[ 4291.697205] usbcore: registered new interface driver uvcvideo
[ 4291.697205] USB Video Class driver (1.1.1)
`
It seems OK now ,but when I ran realsense-viewer chosen code-light depth sensor I still got nothing and had 
`xioctl(VIDIOC_DQBUF) failed Last Error: No such device`

GITSHOHOKU, 2018-03-01:
And when I did patch again and ran realsense-viewer code-light depth sensor I got 
```
Backend in rs2_open_multiple(sensor:0x7f5b08000bf0, profiles:0x7f5b08025de0, count:1):
The requested pixel format 'INZI' is not natively supported by the Linux kernel and likely requires a patch!
Alternatively please upgrade to kernel 4.12 or later. Last Error: Inappropriate ioctl for device
```

closed 2018-03-01 by GITSHOHOKU

bigbellmercy, 2020-06-23:
> * Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):
>
>   * Consider checking out SDK [...]

Additional information for the problem "modprobe: ERROR: could not insert 'videodev':
**After Secure Boot disabled in computer BIOS setup, the problem disappeared.**
[Related issue](https://github.com/IntelRealSense/librealsense/issues/5598)

The succeeded result of the script ./scripts/patch-realsense-ubuntu-lts.sh:
```
Patched kernels modules were created successfully

0
Replacing videodev  -
	Applying the patched module ...  succeeded
Replacing uvcvideo  -
	Applying the patched module ...  succeeded
Replacing hid_sensor_accel_3d  -
	Applying the patched module ...  succeeded
Replacing hid_sensor_gyro_3d  -
	Applying the patched module ...  succeeded

Script has completed. Please consult the installation guide for further instruction.

```

adricort, 2024-03-17:
> Thanks for your reply ev-mp ,I ues `sudo apt-get install realsense-uvcvideo`to install kernel patches but I got error again uvcvideo: Running module version sanity check.
>
>     * Original module [...]

In my case, the Secure Boot also solved my problem! Thanks!