realsenseai/librealsense #1228: cannot start realsense with roslaunch

agari1229, 2018-02-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |{D400 } | 
| Firmware Version                   || 
| Operating System & Version |   {Ubuntu16.04LTS | 
| Kernel Version (Linux Only)    | 4.13.0-36-generic                                         | 
| Platform                                 | PC|
| SDK Version                            |  { legacy / 2.10.0 }                          | 

### Issue Description
The result from 
`roslaunch realsense_ros_camera rs_camera.launch`
is
`...[ERROR] [1519547419.761483128]: An exception has been thrown: libusb_bulk_transfer(...) returned LIBUSB_ERROR_IO Last Error: Resource temporarily unavailable
[FATAL] [1519547419.761874994]: Failed to load nodelet '/camera/realsense_ros_camera` of type `realsense_ros_camera/RealSenseNodeFactory` to manager `realsense_ros_camera_manager'
[camera/realsense_ros_camera-3] process has died [pid 2139, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense_ros_camera/RealSenseNodeFactory realsense_ros_camera_manager __name:=realsense_ros_camera __log:=/home/yuki/.ros/log/1c2f9e16-1a06-11e8-a924-d4258b2da0d7/camera-realsense_ros_camera-3.log].
log file: /home/yuki/.ros/log/1c2f9e16-1a06-11e8-a924-d4258b2da0d7/camera-realsense_ros_camera-3*.log
^C[camera/realsense_ros_camera_manager-2] killing on exit
[ INFO] [1519547422.255184747]: Interrupt Signal is received! Terminating RealSense Node...
[camera/realsense_ros_camera_manager-2] escalating to SIGTERM
`

i used `sudo chmod 666 /dev/video...`commando, but there was no change.

icarpis, 2018-02-25:
Hello @agari1229, have you managed to streaming using LibRealSense tools such as [realsense-viewer](https://github.com/IntelRealSense/librealsense/tree/master/tools/realsense-viewer) or [rs-capture](https://github.com/IntelRealSense/librealsense/tree/master/examples/capture)?
Also, could you please provide the output of `rs-enumerate-devices`?

agari1229, 2018-02-25:
i could solve the problem by applying the patch of uvcvideo. 
i referred ti the following two. 
[url1](http://wiki.ros.org/librealsense)
[url2](http://asukiaaa.blogspot.jp/2016/05/ubuntu1404librealsenserealsensesr300.htmll)

Thanks.

closed 2018-02-26 by jlgarcia75