realsenseai/librealsense #1229: How to load config json file ? And there are any other way to configure my camera ?

zenving, 2018-02-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Ubuntu 16  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

How to load config json file ?
And I want to know more way to config my D415 to work better,  I use realsenViewer tools and save  the config to json file. But I can't use the API to load the json file in my code. Please help me, thank you.

dorodnic, 2018-02-25:
Hi @zenving 

Please add the following includes:
```cpp
#include <librealsense2/rs_advanced_mode.hpp>
#include <string>
#include <fstream>
#include <streambuf>
```
And then when you configure the pipeline, add:
```cpp
auto profile = pipe.start();
std::ifstream t("test.json");
std::string str((std::istreambuf_iterator<char>(t)),
                std::istreambuf_iterator<char>());
rs400::advanced_mode dev = profile.get_device();
dev.load_json(str);
```
Very similar API is also available in C (and python). 

Alternatively, you can access the individual sensors and manually modify all the options and advanced settings. JSON files are just a convenient way to bundle everything together.

zenving, 2018-02-25:
Hi @dorodnic ,thanks for your help, I changed my code according your reply. But there are some things looks wrong. 
First, here are my json file:
{
    "aux-param-autoexposure-setpoint": "400",
    "aux-param-colorcorrection1": "0.129883",
    "aux-param-colorcorrection10": "-0.441406",
    "aux-param-colorcorrection11": "-0.441406",
    "aux-param-colorcorrection12": "-0.0390625",
    "aux-param-colorcorrection2": "0.399414",
    "aux-param-colorcorrection3": "0.399414",
    "aux-param-colorcorrection4": "-0.0693359",
    "aux-param-colorcorrection5": "-0.198242",
    "aux-param-colorcorrection6": "-0.40332",
    "aux-param-colorcorrection7": "-0.40332",
    "aux-param-colorcorrection8": "1.00586",
    "aux-param-colorcorrection9": "0.921875",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "False",
    "controls-autoexposure-manual": "12220",
    "controls-color-autoexposure-auto": "False",
    "controls-color-autoexposure-manual": "166",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "30",
    "controls-depth-white-balance-auto": "False",
    "controls-laserpower": "150",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-autoexposure-setpoint": "400",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-colorcorrection1": "0.129883",
    "param-colorcorrection10": "-0.441406",
    "param-colorcorrection11": "-0.441406",
    "param-colorcorrection12": "-0.0390625",
    "param-colorcorrection2": "0.399414",
    "param-colorcorrection3": "0.399414",
    "param-colorcorrection4": "-0.0693359",
    "param-colorcorrection5": "-0.198242",
    "param-colorcorrection6": "-0.40332",
    "param-colorcorrection7": "-0.40332",
    "param-colorcorrection8": "1.00586",
    "param-colorcorrection9": "0.921875",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "1",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "751",
    "param-lambdacensus": "6",
    "param-leftrightthreshold": "10",
    "param-maxscorethreshb": "2893",
    "param-medianthreshold": "796",
    "param-minscorethresha": "4",
    "param-neighborthresh": "108",
    "param-raumine": "6",
    "param-rauminn": "3",
    "param-rauminnssum": "7",
    "param-raumins": "2",
    "param-rauminw": "2",
    "param-rauminwesum": "12",
    "param-regioncolorthresholdb": "0.785714",
    "param-regioncolorthresholdg": "0.565558",
    "param-regioncolorthresholdr": "0.985323",
    "param-regionshrinku": "3",
    "param-regionshrinkv": "0",
    "param-robbinsmonrodecrement": "25",
    "param-robbinsmonroincrement": "2",
    "param-rsmdiffthreshold": "1.65625",
    "param-rsmrauslodiffthreshold": "0.71875",
    "param-rsmremovethreshold": "0.809524",
    "param-scanlineedgetaub": "13",
    "param-scanlineedgetaug": "15",
    "param-scanlineedgetaur": "30",
    "param-scanlinep1": "155",
    "param-scanlinep1onediscon": "160",
    "param-scanlinep1twodiscon": "59",
    "param-scanlinep2": "190",
    "param-scanlinep2onediscon": "507",
    "param-scanlinep2twodiscon": "493",
    "param-secondpeakdelta": "647",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "1722",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-depth-format": "Z16",
    "stream-fps": "90",
    "stream-height": "480",
    "stream-width": "848"
}
The output of the config's depth frame are looks same as in  realsense-viewer-tools, where I get the json file.　But, the stream-fps and frame's size is not my want, not the setting I want  in the json file,

zenving, 2018-02-25:
Hi, @dorodnic  and I find a interesting thing, that when I changed my "realsenseviewer" gui tools 's setting, and then, I just use a sample pipeline, not load the json file, I find the output of the stream's frame is same as the realsenseViewer tools  where I setting before, it same the setting is saved, but I not load the setting in my code, I want to know how it works, is it saved in the camera inside ? Please help me, thx.

dorodnic, 2018-02-25:
I believe the settings will persists as long as camera is connected. 
Currently, the method I mentioned will ignore resolution and FPS settings, since these are already set by the pipeline. The viewer is doing this in two steps - first it reads FPS and resolution from JSON and configures the camera, then loads the rest with `advanced_mode` class. 
We plan to add a method to the pipeline that would allow to start from JSON, that would include FPS and resolution.

zenving, 2018-02-26:
Thanks first ! But how can I set pipeline to make the device be configure with json file, also have the FPS and resolution I need at the same time. I try to use rs2::config to set the FPS and resoultion, and then accoriding to your method, but there are some wrong like this: 

terminate called after throwing an instance of 'rs2::error'
  what():  null pointer passed for argument "dev"

and my some code show in here:

```
   rs2::pipeline pipe;
   rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH, 848, 480, RS2_FORMAT_Z16, 90); 
    pipe.start(cfg);
    auto profile = pipe.get_active_profile();
    std::ifstream t("/home/ving/test.json");
    std::string str((std::istreambuf_iterator<char>(t)),
                    std::istreambuf_iterator<char>());
    rs400::advanced_mode dev = profile.get_device();
    dev.load_json(str);
```
What's more I want to know more clearly in your reply " The viewer is doing this in two steps - first it reads FPS and resolution from JSON and configures the camera, then loads the rest with advanced_mode class. " What is the "viewer "? Is the realsenseVIewer GUI tool ? and what's "rest with advanced_mode class" real mean, is it not same as the setting in the json file ? 
Please help me, thank you !!!

dorodnic, 2018-02-26:
I'm referring to the following code: [model-views.cpp:4867](https://github.com/IntelRealSense/librealsense/blob/master/common/model-views.cpp#L4867) used by both the RealSense Viewer and the Depth Quality Tool. 
First, we set all the controls using `advanced.load_json(str);`, then [device_model::load_viewer_configurations()](https://github.com/IntelRealSense/librealsense/blob/master/common/model-views.cpp#L4709) configures FPS and resolution. 

Could you clarify if the code snippet worked for you or did you get an empty `dev`?

zenving, 2018-02-26:
I get an empty dev...when use the code snippet 
  ```
 rs2::pipeline pipe;
   rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH, 848, 480, RS2_FORMAT_Z16, 90); 
    pipe.start(cfg);
    auto profile = pipe.get_active_profile();
    std::ifstream t("/home/ving/test.json");
    std::string str((std::istreambuf_iterator<char>(t)),
                    std::istreambuf_iterator<char>());
    rs400::advanced_mode dev = profile.get_device();
    dev.load_json(str);
```

But if I start pipe closed the cfg start like 
`pipe.start();`it works, but the fps and resolution not my want.

dorodnic, 2018-02-26:
Do presets work on the same device? Could you please check the firmware version in the RealSense Viewer (click on i button to see)?

zenving, 2018-02-26:
Yes, the same device.
librealsense version: 2.9.1
firmware version:05.08.15.00

dorodnic, 2018-02-26:
This seems to be a bug. We currently don't support recording advanced mode settings into ROS-bag, but this should not prevent you from using the device while recording.
I created a pull-request to address it.

closed 2018-03-11 by zenving