realsenseai/librealsense #1230: Python: Alignment of depth and RGB causes lines along depth dimension

dolhasz, 2018-02-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC Python |
| SDK Version                            |  2 }                          | 

### Issue Description
Hi there, I have notced that some errors are introduced into the depth frame data when RGB and Depth alignment is performed. 

In the first example on the left I am taking the depth data and running it through the rs.colorizer.

The second example is exactly the same process, but I take an aligned depth frame (output of rs.align). 

Notice the horizontal lines appearing along the depth direction. Is this normal behaviour? It is causing a lot of issues with processing my depth data down the line.

Please let me know if there is a way of fixing this.

See this link for visual example of the issue: https://imgur.com/1cO9cKz

Thanks,

Alan

RealSense-Customer-Engineering, 2018-03-13:
[Realsense Customer Engineering Team Comment] 
@dolhasz  Do you want to do "depth align to color" or "color align to depth"? The sample "rs-align.cpp" is for "depth align to color". Seems you're not using the original sample codes. Could you please provide a code snippet?

RealSense-Customer-Engineering, 2018-04-11:
[Realsense Customer Engineering Team Comment] 
@dolhasz Please update if you still need support. Thanks!

RealSense-Customer-Engineering, 2018-04-18:
[Realsense Customer Engineering Team Comment] 
@dolhasz Didn't get your response yet. Will close this ticket. If you still get the issue, please create a new one. Thanks!

closed 2018-04-18 by RealSense-Customer-Engineering