realsenseai/librealsense #1232: Ｗhere I can find the function prototype "pointcloud.calculate(depth);"

zenving, 2018-02-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Ubuntu 16  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
In the "pointcloud" example :

 // Generate the pointcloud and texture mappings
        points = pc.calculate(depth);

I what to find the "calculate" function prototype, but I cant. I want to know is it as the same as the function in the headfile "rsutil.h" ?

dorodnic, 2018-02-26:
Hi @zenving 
`pointcloud::calculate` calls [pointcloud::process_depth_frame](https://github.com/IntelRealSense/librealsense/blob/master/src/proc/pointcloud.cpp#L135). 
Right now it is just calling `rsutils.h` in a loop. The point of abstracting away `pointcloud` and `align` in a class is to allow us to change this in the future. In particular, there are two alternative implementations of `process_depth_frame` in R&D - one using SIMD and one using CUDA.

zenving, 2018-02-27:
Ｈi, @dorodnic, do you have some documents about how to setting the D415 camera's paramters in the reaslenseViewer pannel to make it work better ?

zenving, 2018-02-27:
Hi, @dorodnic. I find a new question that I can't get the intrinsics's coeffs[] ...
Here is my some code:

```
        cfg.enable_stream(RS2_STREAM_DEPTH, 848, 480, RS2_FORMAT_Z16, 90); 
        cfg.enable_stream(RS2_STREAM_COLOR, 848, 480,   RS2_FORMAT_BGR8, 30);
        cfg.enable_stream(RS2_STREAM_INFRARED);
         pipe.start(cfg);
         auto profile = pipe.get_active_profile();
         auto stream = profile.get_stream( RS2_STREAM_DEPTH ).as<rs2::video_stream_profile>();;
         rs2_intrinsics intrins;
         intrins = stream.get_intrinsics();
        cout << "model = " << intrins.model << endl;
        cout << "coeffs[0] = " << intrins.coeffs[0] << endl;
        cout << "coeffs[1] = " << intrins.coeffs[1] << endl;
        cout << "coeffs[2] = " << intrins.coeffs[2] << endl;
        cout << "coeffs[3] = " << intrins.coeffs[3] << endl;
        cout << "coeffs[4] = " << intrins.coeffs[4] << endl;
        cout << "weight = " << intrins.width << endl;
        cout << "hight = " << intrins.height <<endl;
        cout << "ppx = " << intrins.ppx << endl;
        cout << "ppy = " << intrins.ppy << endl;
        cout << "fx = " << intrins.fx << endl;
        cout << " fy = " << intrins.fy << endl;
```

but the out put of the coeffs are 0, here is the output 

```
model = Brown Conrady
coeffs[0] = 0
coeffs[1] = 0
coeffs[2] = 0
coeffs[3] = 0
coeffs[4] = 0
weight = 848
hight = 480
ppx = 428.256
ppy = 242.019
fx = 628.689
 fy = 628.689
model = Brown Conrady
coeffs[0] = 0
```

dorodnic, 2018-03-01:
[D400-Series-Visual-Presets](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets) has some comments and links related to acquisition parameters. I can also recommend to look at some of white papers available at [realsense.intel.com](https://realsense.intel.com/downloads/).
Regarding RGB coefficient, this result is the expected behavior. The distortion of the RGB sensor far from center is not that big, nearly negligible for real life applications. That's why you can estimate RGB projection as simple (x/z, y/z). 
Hope this help

zenving, 2018-03-01:
@dorodnic Thanks.

closed 2018-03-02 by jlgarcia75