realsenseai/librealsense #1235: Quality of masured depth data

UweDorsch, 2018-02-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Intel RealSense 435                                | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Win 10 Pro 64  | 
| Platform                                 | PC |
| SDK Version                            | 2.10.0                          | 

### Quality of masured depth data
During my first tests with the D435 Camera I observed that the measured depth points always building smooth surfaces. There is no noise in the depth data if you look at a smal selection of points. But you can see that there are structures like little waves in the depth data. It looks like some preprocessing to denoise the data, but this effect looks like there is a structure in a flat surface. It would be helpfull to get the raw data with all the noise and then I can look what points I would use or not.

I tried to use different presets, the high accuracy delivers the best results (smalest waves), but I did not find anything to get depth data like I get from Kinect or Primsense.
I think that post processing the output is not the solution, I just like to get a less preprocessed set of depth data. The problem is getting worse with further distance (>1,5m).
I would be happy to get some hints.

There is an issue with a similar problem especial for smal distances:
Finding the best acquisition parameters (D415) (https://github.com/IntelRealSense/librealsense/issues/1171)

jlgarcia75, 2018-02-26:
Hello @UweDorsch,

You may find this [discussion](https://communities.intel.com/thread/121880) on the RealSense Community helpful.

Regards,
Jesus G.
Intel Customer Support

UweDorsch, 2018-02-27:
Hello Jesus,
thank you for your response.
The discussion on the RealSense Community is exactly about my problem. But it is a little bit disappointing, because it looks like there is no way to get depth data from the RealSense Camera with comparable quality to the kinect depht data. 
The wavy depth values sems to be the result of the internal depth processing inside the RealSense Camera. Is there a opportunity to get less preprocessed data?

kind regards
Uwe

RealSense-Customer-Engineering, 2018-04-13:
[Realsense Customer Engineering Team Comment] 
Hello @UweDorsch,

Have you seen the RealSense Tuning whitepaper? https://realsense.intel.com/wp-content/uploads/sites/63/BKMs-For-Tuning-RealSense_D4xx_Cameras_WP_1.7.pdf

Regards,
Jesus G.
Intel Customer Support

closed 2018-05-11 by RealSense-Customer-Engineering

jlgarcia75, 2018-07-05:
Hello @ruktech,

You can find the tuning document here:https://www.intel.com/content/www/us/en/support/articles/000027833/emerging-technologies/intel-realsense-technology.html.

Regards,
Jesus
Intel Customer Support