realsenseai/librealsense #1237: Patch for compilation error for non-optimized builds

richardroberts, 2018-02-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | N/A | 
| Operating System & Version |   Clang, Linux, Debian-based  | 
| Kernel Version (Linux Only)    |  N/A                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.9.0                         | 

### Issue Description
In a non-optimized build, the following compilation error occurs.  It seems to be due to a call to sort() within src/sync.cpp.  The custom comparator should have a const reference arguments according to C++ standard library requirements, but the provided one has non-constant reference arguments.  I'm not sure why this only occurs in non-optimized builds... perhaps the function is optimized away otherwise since we're not using it?

The attached patch fixes this by making the comparator function use const reference arguments:
[realsense_non_optimized_compilation_fix.patch.txt](https://github.com/IntelRealSense/librealsense/files/1760617/realsense_non_optimized_compilation_fix.patch.txt)

Compilation error:

```
In file included from third_party/librealsense2/librealsense-2.9.0/src/sync.cpp:4:
In file included from third_party/librealsense2/librealsense-2.9.0/src/proc/synthetic-stream.h:6:
In file included from third_party/librealsense2/librealsense-2.9.0/src/core/processing.h:5:
In file included from third_party/librealsense2/librealsense-2.9.0/src/core/streaming.h:5:
In file included from third_party/librealsense2/librealsense-2.

9.0/src/core/options.h:8:
In file included from third_party/librealsense2/librealsense-2.9.0/src/types.h:23:
In file included from third_party/stl/gcc3/algorithm:12:
In file included from third_party/crosstool/v18/stable/toolchain/bin/../lib/gcc/x86_64-grtev4-linux-gnu/4.9.x-google/../../../../x86_64-grtev4-linux-gnu/include/c++/4.9.x-google/algorithm:62:
third_party/crosstool/v18/stable/toolchain/bin/../lib/gcc/x86_64-grtev4-linux-gnu/4.9.x-google/../../../../x86_64-grtev4-linux-gnu/include/c++/4.9.x-google/bits/stl_algo.h:214:11: error: no matching function for call to object of type '(lambda at third_party/librealsense2/librealsense-2.9.0/src/sync.cpp:323:55)'
      if (_M_compare(__x, __x))
          ^~~~~~~~~~
third_party/crosstool/v18/stable/toolchain/bin/../lib/gcc/x86_64-grtev4-linux-gnu/4.9.x-google/../../../../x86_64-grtev4-linux-gnu/include/c++/4.9.x-google/bits/predefined_ops.h:141:16: note: in instantiation of function template specialization 'std::_CheckedCompare<(lambda at third_party/librealsense2/librealsense-2.9.0/src/sync.cpp:323:55)>::operator()<librealsense::frame_holder>' requested here
        { return bool(_M_comp(*__it, __val)); }
                      ^
third_party/crosstool/v18/stable/toolchain/bin/../lib/gcc/x86_64-grtev4-linux-gnu/4.9.x-google/../../../../x86_64-grtev4-linux-gnu/include/c++/4.9.x-google/bits/stl_heap.h:129:42: note: in instantiation of function template specialization '__gnu_cxx::__ops::_Iter_comp_val<std::_CheckedCompare<(lambda at third_party/librealsense2/librealsense-2.9.0/src/sync.cpp:323:55)> >::operator()<__gnu_cxx::__normal_iterator<librealsense::frame_holder *, std::vector<librealsense::frame_holder, std::allocator<librealsense::frame_holder> > >, librealsense::frame_holder>' requested here
      while (__holeIndex > __topIndex && __comp(__first + __parent, __value))
                                         ^
third_party/crosstool/v18/stable/toolchain/bin/../lib/gcc/x86_64-grtev4-linux-gnu/4.9.x-google/../../../../x86_64-grtev4-linux-gnu/include/c++/4.9.x-google/bits/stl_heap.h:228:12: note: in instantiation of function template specialization 'std::__push_heap<__gnu_cxx::__normal_iterator<librealsense::frame_holder *, std::vector<librealsense::frame_holder, std::allocator<librealsense::frame_holder> > >, long, librealsense::frame_holder, __gnu_cxx::__ops::_Iter_comp_val<std::_CheckedCompare<(lambda at third_party/librealsense2/librealsense-2.9.0/src/sync.cpp:323:55)> > >' requested here
      std::__push_heap(__first, __holeIndex, __topIndex, 
           ^
third_party/crosstool/v18/stable/toolchain/bin/../lib/gcc/x86_64-grtev4-linux-gnu/4.9.x-google/../../../../x86_64-grtev4-linux-gnu/include/c++/4.9.x-google/bits/stl_heap.h:333:9: note: in instantiation of function template specialization 'std::__adjust_heap<__gnu_cxx::__normal_iterator<librealsense::frame_holder *, std::vector<librealsense::frame_holder, std::allocator<librealsense::frame_holder> > >, long, librealsense::frame_holder, __gnu_cxx::__ops::_Iter_comp_iter<std::_CheckedCompare<(lambda at third_party/librealsense2/librealsense-2.9.0/src/sync.cpp:323:55)> > >' requested here
          std::__adjust_heap(__first, __parent, __len, _GLIBCXX_MOVE(__value),
               ^
third_party/crosstool/v18/stable/toolchain/bin/../lib/gcc/x86_64-grtev4-linux-gnu/4.9.x-google/../../../../x86_64-grtev4-linux-gnu/include/c++/4.9.x-google/bits/stl_algo.h:1707:12: note: in instantiation of function template specialization 'std::__make_heap<__gnu_cxx::__normal_iterator<librealsense::frame_holder *, std::vector<librealsense::frame_holder, std::allocator<librealsense::frame_holder> > >, __gnu_cxx::__ops::_Iter_comp_iter<std::_CheckedCompare<(lambda at third_party/librealsense2/librealsense-2.9.0/src/sync.cpp:323:55)> > >' requested here
      std::__make_heap(__first, __middle, __comp);
           ^
third_party/crosstool/v18/stable/toolchain/bin/../lib/gcc/x86_64-grtev4-linux-gnu/4.9.x-google/../../../../x86_64-grtev4-linux-gnu/include/c++/4.9.x-google/bits/stl_algo.h:1966:12: note: in instantiation of function template specialization 'std::__heap_select<__gnu_cxx::__normal_iterator<librealsense::frame_holder *, std::vector<librealsense::frame_holder, std::allocator<librealsense::frame_holder> > >, __gnu_cxx::__ops::_Iter_comp_iter<std::_CheckedCompare<(lambda at third_party/librealsense2/librealsense-2.9.0/src/sync.cpp:323:55)> > >' requested here
      std::__heap_select(__first, __middle, __last, __comp);
           ^
third_party/crosstool/v18/stable/toolchain/bin/../lib/gcc/x86_64-grtev4-linux-gnu/4.9.x-google/../../../../x86_64-grtev4-linux-gnu/include/c++/4.9.x-google/bits/stl_algo.h:1981:13: note: in instantiation of function template specialization 'std::__partial_sort<__gnu_cxx::__normal_iterator<librealsense::frame_holder *, std::vector<librealsense::frame_holder, std::allocator<librealsense::frame_holder> > >, __gnu_cxx::__ops::_Iter_comp_iter<std::_CheckedCompare<(lambda at third_party/librealsense2/librealsense-2.9.0/src/sync.cpp:323:55)> > >' requested here
              std::__partial_sort(__first, __last, __last, __comp);
                   ^
third_party/crosstool/v18/stable/toolchain/bin/../lib/gcc/x86_64-grtev4-linux-gnu/4.9.x-google/../../../../x86_64-grtev4-linux-gnu/include/c++/4.9.x-google/bits/stl_algo.h:2001:9: note: in instantiation of function template specialization 'std::__introsort_loop<__gnu_cxx::__normal_iterator<librealsense::frame_holder *, std::vector<librealsense::frame_holder, std::allocator<librealsense::frame_holder> > >, long, __gnu_cxx::__ops::_Iter_comp_iter<std::_CheckedCompare<(lambda at third_party/librealsense2/librealsense-2.9.0/src/sync.cpp:323:55)> > >' requested here
          std::__introsort_loop(__first, __last,
               ^
third_party/crosstool/v18/stable/toolchain/bin/../lib/gcc/x86_64-grtev4-linux-gnu/4.9.x-google/../../../../x86_64-grtev4-linux-gnu/include/c++/4.9.x-google/bits/stl_algo.h:4751:12: note: in instantiation of function template specialization 'std::__sort<__gnu_cxx::__normal_iterator<librealsense::frame_holder *, std::vector<librealsense::frame_holder, std::allocator<librealsense::frame_holder> > >, __gnu_cxx::__ops::_Iter_comp_iter<std::_CheckedCompare<(lambda at third_party/librealsense2/librealsense-2.9.0/src/sync.cpp:323:55)> > >' requested here
      std::__sort(__first, __last, __gnu_cxx::__ops::__iter_comp_iter(__CheckedCompare(__comp)));
           ^
third_party/librealsense2/librealsense-2.9.0/src/sync.cpp:323:22: note: in instantiation of function template specialization 'std::sort<__gnu_cxx::__normal_iterator<librealsense::frame_holder *, std::vector<librealsense::f [... cut ...] _Tp = librealsense::frame_holder]
    bool operator()(const _Tp& __x, const _Tp& __y) {
         ^
third_party/crosstool/v18/stable/toolchain/bin/../lib/gcc/x86_64-grtev4-linux-gnu/4.9.x-google/../../../../x86_64-grtev4-linux-gnu/include/c++/4.9.x-google/bits/stl_algo.h:226:10: note: candidate template ignored: substitution failure [with _Tp1 = librealsense::frame_holder, _Tp2 = librealsense::frame_holder]
    bool operator()(const _Tp1& __x, const _Tp2& __y) {
         ^
5 errors generated.
```

dorodnic, 2018-02-27:
Hi @richardroberts 
Thanks for bringing this up. What version of gcc are you using? Our CI is running with gcc 5.

richardroberts, 2018-03-02:
This is using clang, and this only version info I'm able to get is that it is "4.9.x".

richardroberts, 2018-04-12:
Thanks for the follow-up.  I don't have access to gcc 5 to test with that, but I'll let you know if the issue reoccurs with clang.  Thanks!

closed 2018-04-15 by dorodnic