realsenseai/librealsense #1238: difference in using pipeline or device stream with low latency frame grab

kyranf, 2018-02-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Linux Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.4.0-112 -lowlatency                                        | 
| Platform                                 | Advantech MIO-5350 |
| SDK Version                            | 2.9.1          | 

### Issue Description
In the RS400 support part of the docs, https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md 

it mentions there is a low latency way of getting a frame as soon as possible, with this:
`// Configure direct callback for new frames:
dev.start([](rs2::frame frame){
    auto pixels = frame.get_data(); // pointer to frame data
});
// The application will be notified as soon as new frames become available.`

I was under the impression we had to use pipeline and pipe.start() function to get it streaming, and poll for frames like the example shown in the API How-to part of the github docs:

`pipe.start();
rs2::frameset frames;
if (pipe.poll_for_frames(&frames))
{
    rs2::frame depth_frame = frames.first(RS2_STREAM_DEPTH);
    depth_frame.get_data();
}`

Is there a better (more complete) example or documentation of using the API with device->start() instead of pipe and pipeline in the new realsense library API?

dorodnic, 2018-02-27:
Hi @kyranf 
Yes, we provide the [sensor control](https://github.com/IntelRealSense/librealsense/tree/master/examples/sensor-control) example to demonstrate how you can access individual sensors and streams. 
That said, empirical data show that the overhead of using the pipeline is negligible (considerably less then the inherent latency in the hardware & OS). The difference comes from two effects:
1. Setting a callback will allow you to run your code on the OS thread, avoiding the cost of switching to the main thread. 
2. When you configure multiple streams, pipeline will delay the faster stream until all frames of the frameset are ready.

kyranf, 2018-02-27:
hi @dorodnic 

Okay, so in the api_how_to.h and the sensor control example program, i see that the rs2::sensor object can be 'opened', 'started' and assign a callback function, 'stopped' and then 'closed'.  Is there an equivalent to 'wait for next frame' with the rs2::sensor or rs2::device interface like there was in the legacy API?

dorodnic, 2018-02-27:
You can create `rs2::frame_queue` object and pass it as (instead of) the callback to start. 
Then you can do `rs2::frame f = q.wait_for_frame()` or `q.poll_for_frame(&f)`.
Frame queues should not be shared between multiple sensors. If you need to listen on multiple sensors, you can either create queue per sensor and poll on each one, or create an `rs2::syncer` object and pass it as a callback to multiple sensors. At this point, however, you will be essentially getting the same results as with the pipeline.

kyranf, 2018-02-27:
Thank you for your assistance. I will close this for now, and re-open if i have any other issues/questions

closed 2018-02-27 by kyranf