realsenseai/librealsense #1240: D415 gets corrupted/incomplete frames after a while

kyranf, 2018-02-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Linux Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.4.0-112 -lowlatency                                        | 
| Platform                                 | Advantech MIO-5350 |
| SDK Version                            | 2.9.1          | 

### Issue Description
I am using the v2.9.1 realsense viewer in linux, and after a 2 hour test period I come back and find that it's spewing out constant errors with corrupted/incomplete frames being received. See screen capture below:

[image]

Notice the temperature of the ASIC and laser projector. Could this be causing these malfunctions?

I also noticed this in my dmesg output, duplicated quite a lot, but not spewing out as much as the errors in the viewer. Not sure if directly related:

`[ 9354.354116] WARNING: CPU: 2 PID: 5156 at /build/linux-HSAA8v/linux-4.4.0/drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd86/0x10f0 [videodev]()
[ 9354.354118] Unknown pixelformat 0x49323159
[ 9354.354120] Modules linked in: snd_hda_codec_hdmi ftdi_sio usbserial intel_telemetry_pltdrv intel_pmc_ipc intel_punit_ipc intel_telemetry_core x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel snd_soc_rt286 snd_soc_rl6347a snd_soc_core aesni_intel aes_x86_64 uvcvideo lrw videobuf2_vmalloc snd_hda_codec_realtek snd_compress snd_hda_codec_generic videobuf2_memops videobuf2_v4l2 ac97_bus gf128mul videobuf2_core glue_helper snd_pcm_dmaengine snd_hda_intel ablk_helper v4l2_common cryptd videodev snd_seq_midi snd_hda_codec snd_seq_midi_event media snd_rawmidi snd_hda_core snd_hwdep mei_me snd_seq snd_pcm shpchp mei snd_seq_device snd_timer rfkill_gpio snd soundcore mac_hid 8250_fintek binfmt_misc parport_pc ppdev lp parport autofs4
[ 9354.354170]  i915_bpo intel_ips drm_kms_helper igb syscopyarea ahci sysfillrect sysimgblt dca fb_sys_fops libahci ptp pps_core drm i2c_algo_bit i2c_hid hid video fjes pinctrl_broxton pinctrl_intel
[ 9354.354189] CPU: 2 PID: 5156 Comm: realsense-viewe Tainted: G        W       4.4.0-112-lowlatency #135-Ubuntu
[ 9354.354190] Hardware name:                  /        , BIOS 5.12 07/19/2017
[ 9354.354193]  0000000000000286 e62f22d630322636 ffff88007053fbf0 ffffffff81406311
[ 9354.354196]  ffff88007053fc38 ffffffffc03e9a78 ffff88007053fc28 ffffffff81083af2
[ 9354.354199]  ffff88007053fda0 0000000000000000 0000000000000000 ffff8801773ca000
[ 9354.354202] Call Trace:
[ 9354.354209]  [<ffffffff81406311>] dump_stack+0x63/0x82
[ 9354.354214]  [<ffffffff81083af2>] warn_slowpath_common+0x82/0xc0
[ 9354.354217]  [<ffffffff81083b8c>] warn_slowpath_fmt+0x5c/0x80
[ 9354.354223]  [<ffffffffc047a47d>] ? uvc_ioctl_enum_fmt+0x9d/0xc0 [uvcvideo]
[ 9354.354233]  [<ffffffffc03d46f6>] v4l_enum_fmt+0xd86/0x10f0 [videodev]
[ 9354.354241]  [<ffffffffc03d52e1>] __video_do_ioctl+0x291/0x310 [videodev]
[ 9354.354249]  [<ffffffffc03d4d96>] video_usercopy+0x336/0x5d0 [videodev]
[ 9354.354256]  [<ffffffffc03d5050>] ? video_ioctl2+0x20/0x20 [videodev]
[ 9354.354261]  [<ffffffff811a6577>] ? lru_cache_add_active_or_unevictable+0x27/0xa0
[ 9354.354264]  [<ffffffff811c85f5>] ? handle_mm_fault+0xd35/0x18b0
[ 9354.354271]  [<ffffffffc03d5045>] video_ioctl2+0x15/0x20 [videodev]
[ 9354.354279]  [<ffffffffc03d06a3>] v4l2_ioctl+0xd3/0xe0 [videodev]
[ 9354.354282]  [<ffffffff8122b31f>] do_vfs_ioctl+0x29f/0x490
[ 9354.354285]  [<ffffffff8123571a>] ? __fget+0x7a/0xb0
[ 9354.354288]  [<ffffffff8122b589>] SyS_ioctl+0x79/0x90
[ 9354.354292]  [<ffffffff81853f2d>] entry_SYSCALL_64_fastpath+0x2b/0xe7
[ 9354.356327] ---[ end trace 35c325e72fbc8fad ]---
[ 9354.407631] ------------[ cut here ]------------
[ 9354.407656] WARNING: CPU: 2 PID: 5156 at /build/linux-HSAA8v/linux-4.4.0/drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd86/0x10f0 [videodev]()
[ 9354.407658] Unknown pixelformat 0x00000000
[ 9354.407659] Modules linked in: snd_hda_codec_hdmi ftdi_sio usbserial intel_telemetry_pltdrv intel_pmc_ipc intel_punit_ipc intel_telemetry_core x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel snd_soc_rt286 snd_soc_rl6347a snd_soc_core aesni_intel aes_x86_64 uvcvideo lrw videobuf2_vmalloc snd_hda_codec_realtek snd_compress snd_hda_codec_generic videobuf2_memops videobuf2_v4l2 ac97_bus gf128mul videobuf2_core glue_helper snd_pcm_dmaengine snd_hda_intel ablk_helper v4l2_common cryptd videodev snd_seq_midi snd_hda_codec snd_seq_midi_event media snd_rawmidi snd_hda_core snd_hwdep mei_me snd_seq snd_pcm shpchp mei snd_seq_device snd_timer rfkill_gpio snd soundcore mac_hid 8250_fintek binfmt_misc parport_pc ppdev lp parport autofs4
[ 9354.407709]  i915_bpo intel_ips drm_kms_helper igb syscopyarea ahci sysfillrect sysimgblt dca fb_sys_fops libahci ptp pps_core drm i2c_algo_bit i2c_hid hid video fjes pinctrl_broxton pinctrl_intel
[ 9354.407728] CPU: 2 PID: 5156 Comm: realsense-viewe Tainted: G        W       4.4.0-112-lowlatency #135-Ubuntu
[ 9354.407730] Hardware name:                  /        , BIOS 5.12 07/19/2017
[ 9354.407733]  0000000000000286 e62f22d630322636 ffff88007053fbf0 ffffffff81406311
[ 9354.407736]  ffff88007053fc38 ffffffffc03e9a78 ffff88007053fc28 ffffffff81083af2
[ 9354.407739]  ffff88007053fda0 0000000000000000 0000000000000000 ffff8801773ca300
[ 9354.407743] Call Trace:
[ 9354.407751]  [<ffffffff81406311>] dump_stack+0x63/0x82
[ 9354.407756]  [<ffffffff81083af2>] warn_slowpath_common+0x82/0xc0
[ 9354.407758]  [<ffffffff81083b8c>] warn_slowpath_fmt+0x5c/0x80
[ 9354.407765]  [<ffffffffc047a47d>] ? uvc_ioctl_enum_fmt+0x9d/0xc0 [uvcvideo]
[ 9354.407775]  [<ffffffffc03d46f6>] v4l_enum_fmt+0xd86/0x10f0 [videodev]
[ 9354.407783]  [<ffffffffc03d52e1>] __video_do_ioctl+0x291/0x310 [videodev]
[ 9354.407787]  [<ffffffff81221a0b>] ? terminate_walk+0x6b/0xe0
[ 9354.407791]  [<ffffffff812416d2>] ? __inode_wait_for_writeback+0x72/0xe0
[ 9354.407799]  [<ffffffffc03d4d96>] video_usercopy+0x336/0x5d0 [videodev]
[ 9354.407807]  [<ffffffffc03d5050>] ? video_ioctl2+0x20/0x20 [videodev]
[ 9354.407815]  [<ffffffffc03d5045>] video_ioctl2+0x15/0x20 [videodev]
[ 9354.407822]  [<ffffffffc03d06a3>] v4l2_ioctl+0xd3/0xe0 [videodev]
[ 9354.407825]  [<ffffffff8122b31f>] do_vfs_ioctl+0x29f/0x490
[ 9354.407828]  [<ffffffff8123571a>] ? __fget+0x7a/0xb0
[ 9354.407831]  [<ffffffff8122b589>] SyS_ioctl+0x79/0x90
[ 9354.407836]  [<ffffffff81853f2d>] entry_SYSCALL_64_fastpath+0x2b/0xe7
[ 9354.407838] ---[ end trace 35c325e72fbc8fae ]---
`

dorodnic, 2018-02-27:
In general, partial frames indicate an issue at the USB sub-system. The USB driver (or controller) are only managing to fetch part of each frame. Is your board actively-cooled? 

We regularly perform long tests with the camera, but don't generally encounter such problems. The only side-effect of the projector overheating I am aware of is that the camera can decide to turn-off the laser (and continue to work based on ambient light)

kyranf, 2018-02-27:
turns out my industrial single board computer is overheating and the intel powerclamp driver is forcing 50% idle on all cores, ruining performance and causing frame drops/corruption. 

This is a hardware issue on my end, closing the issue. thanks and sorry for the distraction!

closed 2018-02-27 by kyranf

akouri-doordash, 2019-06-03:
I am having this same issue, but it starts right away on my setup (Lenovo Laptop on Ubuntu 18.04)

kyranf, 2019-06-04:
@akouri-doordash   are you getting the camera detected on USB3 properly? Or is it showing up as a USB2 device?

akouri-doordash, 2019-06-05:
It shows up as a USB2 device. I think it might have something to do with the linux drivers on the laptop because when I boot into Windows it works fine (no frames dropped/partial frames).

kyranf, 2019-06-05:
The libusb drivers and for the UVC drivers have to be kernel patched depending on your Linux OS version and kernel it runs on. What are those? I got mine working with Ubuntu

akouri-doordash, 2019-06-05:
I am working with Ubuntu 18.04. Can you point me to a link on how to patch them?

kyranf, 2019-06-05:
I know there are patches for UVC for 16.04 LTS.. if you look in the librealsense sources there are loads of patches scripts and automated kernel fixing routines. Some work, some don't. It was a bunch for trial and error for me.