realsenseai/librealsense #1242: About latency ...

zenving, 2018-02-27:
| Camera Model                       |  D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
Hello, I find a word in https://github.com/IntelRealSense/librealsense/wiki/Release-Notes.
there are one known issue "Latency of 100ms (DSO-7745)", but how can I fix that, and how can I get every frame's accurate time ? is the time based the hardwaire, or based my computer ?

dorodnic, 2018-02-27:
Hi @zenving 
"Latency of 100ms" is somewhat of a code-name we use for the ongoing latency investigations and improvements. In practice, different sensors and configurations produce different results. We offer public `latency-tool` under OpenCV examples that can provide some rough estimates for some streams. 
Currently, the RGB stream suffers from most latency, with about 100-140ms on Windows. IR and depth stream have less latency, about 60-70ms. Based on our measurements, Linux driver stack has less inherent latency than the Microsoft one.
Providing more optimized color conversions, projection and colorizer code would also reduce several milliseconds from the overall time. 
We are also looking into ways to improve these numbers on firmware level.

dorodnic, 2018-02-27:
You can get hardware timestamp using `frame::get_timestamp()` method. Please make sure `frame::get_frame_timestamp_domain()` returns `RS2_TIMESTAMP_DOMAIN_HARDWARE_CLOCK`.
You can also fetch host time the frame reached librealsense using `frame.get_frame_metadata(RS2_FRAME_METADATA_TIME_OF_ARRIVAL)` and when it reached the driver (Media Foundation / V4L2) using `frame.get_frame_metadata(RS2_FRAME_METADATA_BACKEND_TIMESTAMP)`. 
Please note that frame timestamp is not comparable to the later two, since there is no clock synchronisation between camera and host clocks.

fsasm, 2018-03-01:
What is the target frame time "Latency of 100ms"? Is it possible to achieve a latency of <40 ms for the depth and IR stream on Linux in the near future?

dorodnic, 2018-03-01:
We are trying to get rid of all the overheads, in firmware, software and the driver and get it as low as possible. It depends on the resolution, but latency of 60ms is a reasonable goal to achieve. From my previous experience, going below 60 becomes challenging when the camera is connected via USB, I might be wrong.

kyranf, 2018-03-06:
does the frame rate of the stream help with the latency? or is the entire pipeline always lagging behind? how consistent is the latency? Have you guys got a benchmark or statistical analysis of the latency? If it's fairly consistent that's good.. but if it varies by +-30->60ms a lot, that's going to be horrible to deal with in a real time vision system :(

closed 2018-03-11 by zenving