realsenseai/librealsense #1243: Installing SDK 2.0 on Windows 7

FirefoxMetzger, 2018-02-27:
I am trying to install the RealSense SDK 2.0 on a Windows 7 64-bit machine. I am compiling the library from source and am building python wrappers for python 3.6. I have it working on Linux (Ubuntu 16.04) and Windows 10.

On my Windows 7 machine I can `include pyrealsense2`, but when I actually try to start a pipeline the application crashes with now error message. I assume this is where the .dll is actually loaded, so there might be something wrong with the compilation.

In the device manager I can see the RealSenseD415 Depth and RGB camera, however it shows:
This device cannot start. (Code 10)

Is Windows 7 not supported? Any pointers as to what could cause this behavior? As it simply crashes without an error message I am a bit clueless where to start looking for errors.

dorodnic, 2018-02-27:
Hi @FirefoxMetzger 
Unfortunately we do not officially support Windows 7.
Personally, I wasn't able to make it work on my laptop, I suspect because Windows 7 UVC driver is not properly handling UVC1.5 device. 
We also invested some time into making it work with libusb and libuvc (bypassing Windows drivers all together), but struggled from one problem to another as it seems that libusb+libuvc combination was never properly tested on Windows.

FirefoxMetzger, 2018-02-27:
Ah, thanks for the information.
I will try the variant with installing libusb tomorrow and see if this works for me.

RealSense-Customer-Engineering, 2018-03-15:
[Realsense Customer Engineering Team Comment] 
hi @FirefoxMetzger,

would like to know how it works, and still need help on this topic?
thanks.