realsenseai/librealsense #1244: Viewer is crashing on just installed D430

GlennMiller, 2018-02-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D400 
| Firmware Version                   | 5.8.15.0  
| Operating System & Version |   Windows 10 
| SDK Version                            |  2.10.0                         

### Issue Description
Exception while opening. It occurs in win-uvc.cpp @ line 682, according to the debugger. Screen shot attached.

[image]

dorodnic, 2018-02-28:
Hello @GlennMiller 
Could you please specify the exact Windows version you are using?
Do you see both depth and RGB RealSense cameras in the Device Manager?
The fact that the `camera_control` seem to be `NULL` indicate some severe issue at the Media Foundation level and should not happen.

GlennMiller, 2018-02-28:
Windows 10 pro. Version 10.0.16299 

I do see both depth and RGB RealSense cameras in the Device Manager.

[image]

RealSense-Customer-Engineering, 2018-03-01:
[Realsense Customer Engineering Team Comment] 
Hi @GlennMiller

You can try to use the pre-built realsense viewer app based on 2.10.0? Not reproduce this issue on my Win10 16299 NB + D435 camera (FW: 5.8.15).
https://github.com/IntelRealSense/librealsense/releases/download/v2.10.0/Intel.RealSense.Viewer.exe

RealSense-Customer-Engineering, 2018-03-01:
[Realsense Customer Engineering Team Comment] 
Hi @GlennMiller

In addition to have RealSense D435 camera connected, any other USB camera connected? I can see the similar EXPOSURE GetRange Error if there is another USB camera which is also connected to the system.

GlennMiller, 2018-03-01:
I had a Kinect connected via USB. Once I disconnected it, the RealSense viewer worked perfectly. Thanks very much. I would prefer both be available to me, but I can live with that for the time being. 

Regards,

Glenn Miller
Aliquippa, PA

closed 2018-03-01 by GlennMiller