realsenseai/librealsense #1251: [Feature Request] Add option to specify CRT linking method (/MD or /MT)

UnaNancyOwen, 2018-02-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| SDK Version                            |  2.x.x                                                       | 

### Issue Description
This is feature request.
Please consider add an option to specify linking method (<code>/MD</code> or <code>/MT</code>) of CRT.
Currently, It is determined settings by library link type (static link library or dynamic link library). We can not choose different methods.

dorodnic, 2018-02-28:
Hi @UnaNancyOwen 
I believe it would be relatively easy to add, if this would be useful for you. 
If I recall correctly, we are hard-coding the CMake for `/MT` to make sure people can get started with RealSense without any annoying prerequisites, this is the only reason.

UnaNancyOwen, 2018-02-28:
@dorodnic Currently, We are working to support librealsense2 (pcl::RealSense2Grabber) on PCL.
In the future, I would like to enable this feature by default in Installer (that contains pre built libraries) for Windows that distribute by official. It needs to bundle static link library of librealsense2, but it is necessary to align the CRT link method with other dependent libraries.
Therefore, I would like to add this option.

I think that the default setting should be left for get started without any annoying prerequisites.
However, I think it is necessary an option to specify linking method (/MD or /MT) of CRT.

UnaNancyOwen, 2018-03-01:
@dorodnic The same options that I proposed are implemented in OpenCV.
In the OpenCV, It is <code>BUILD_WITH_STATIC_CRT</code> ([CMakeLists.txt#L295](https://github.com/opencv/opencv/blob/3.4.1/CMakeLists.txt#L295)).

PR #1262 (2018-03-04): Add BUILD_WITH_STATIC_CRT option
Add <code>BUILD_WITH_STATIC_CRT</code> option.
If this option is set to <code>ON</code>, CRT is statically linked.

Closes #1251
closed 2018-03-05 by UnaNancyOwen