realsenseai/librealsense #1252: Estimated error depending on distance to a plane?

TheMikeyR, 2018-02-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 / D435 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.14.13-041413-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  v2.10.0                          | 

### Issue Description
Hello, I'm wondering if there are any information on how big of an error the D400-series have depending on the distance, I know similar charts have been made for the earlier realsense cameras. Here is a test conducted for the [SR300](http://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=7929364.

agrunnet, 2018-02-28:
It is always a bit controversial to post data about performance as there are many caveats. However, for the RealSense cameras looking at a textured wall, and measuring the local RMS noise (fit-to-plane) you should expect something like this chart, with some variability from unit to unit. 
[image]

The D415 should be operated at 1280x720 and has a baseline of 55mm. It has 2x 2MP Color rolling shutter CMOS sensors for stereo. The horizontal FOV is 65 deg. The minimum distance at this resolution is 44cm.  You can reduce the minZ by reducing imager resolution. 

The D435 should be operated at 848x480 and has a baseline of 50mm. It uses 2x 1MP Monochrome global imagers for stereo. The horizontal FOV is 90 deg.  The minimum distance at this resolution is 17cm. (With smaller resolution it will be shorter). 

So as you can see, the D415 is expected to have >2x better depth at a given distance.   
 
With good object texture you will generally get best results without laser projector turned on. However, when there is a lack of texture (like a white wall) or low texture, you will definitely need the projector. 
Also, there is some unit-to-unit variability of lens distortion correction and calibration, so you may not see this number when you measure across the whole field-of-view. Also, this is not the ABSOLUTE error. 

Note that this graph should be used merely as a guideline to compare the two different models and get a rough idea of what you could expect.   

The units are focused at 50cm, so for near operation (~<20cm) the performance deviates a bit from this due to imagers being out of focus. 

In general the RealSense imagers love sunlight, because there more light the better the stereo images. In principle the max range of the RealSense cameras is >65m on textured targets.  However, indoors in office lighting the IR projector SNR diminishes after about 3m. You can mitigate this by putting IR-bandpass filters on the stereo imagers or using an external projector. 

For an external projector, you can use any lighting projector as there is no bandpass filter on the imagers. So visible to 850nm to 940nm all work. It is important to reduce laser speckle if you are using a laser-based projector. In general any semi-random pattern works great, but try to have more than 5K and less than 300K dots. The stereo resolution will be more than this because it does not just derive depth from where the projected dots are. 

I hope this gives some guidance that is helpful.

TheMikeyR, 2018-03-01:
Thanks a lot for the insight it is definitely a great help!

closed 2018-03-01 by TheMikeyR