realsenseai/librealsense #1253: Kernel issue with librealsense with Ubuntu 14 on UP Board

littlejohnyang, 2018-03-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 14)  | 
| Kernel Version (Linux Only)    |  4.4.0-2-upboard                            | 
| Platform                                 | UP board  |
| SDK Version                            |  { 2 }                          | 

### Issue Description
I tried to run the script:
./scripts/patch-realsense-ubuntu-xenial.sh
but ended up getting this error:
sudo modprobe uvcvideo produces dmesg: uvc kernel module is not loaded

I think before running the script, I can actually detect have uvcvideo running fine showing in dmesg. Is there anyway to revert the patch or fix it through kernel support for 4.4.0-2-upboard (which is the latest support for ubuntu 14 on UP board)

RealSense-Customer-Engineering, 2018-03-01:
[Realsense Customer Engineering Team Comment] 
Hi @littlejohnyang

In addition to modify the patch, do you also consider to use Ubuntu 16.04 on the UP board? Then, you can try to install the uvcvideo modules provided by DKMS as below link.
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

littlejohnyang, 2018-03-01:
I actually got it working now after reformat the whole PC and install from scratch again. What happened is that the ./scripts/patch-realsense-ubuntu-xenial.sh patch causes all the problem. So I can now get all the images without running that script. However, I am getting the following warning constantly even though everything is working fine:
Failed to read busnum/devnum. Device Path: /sys/bus/iio/devices/iio:device0

Anyone know how to solve the problem with this warning message?

Also from dmesg, I can see a message saying:
[16473.523394] uvcvideo: Unable to create debugfs 2-2 directory.

I'm not sure if this is going to be a problem. Otherwise it seems working fine except the constantly printed warning message.

RealSense-Customer-Engineering, 2018-03-01:
[Realsense Customer Engineering Team Comment] 
Hi @littlejohnyang,

Do you still use Ubuntu 14.04 + kernel 4.4.0-2? Suggest to have below patches for hid and camera formats, etc.
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
$ sudo apt-get install realsense-uvcvideo

Then, you can check if the warning message "Failed to read busnum/devnum" still there.

littlejohnyang, 2018-03-01:
Somehow the realsense-uvcvideo isn't available for ubuntu14 I think. However, after a system reboot yesterday, the "Failed to read busnum/devnum" warning message disappeared now. However, it'll throw the following warning and error message despite the camera is still functioning correctly.

 02/03 10:21:17,276 WARNING [140117030401792] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1519946477.295799525]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 02/03 10:21:17,343 WARNING [140117006227200] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
 02/03 10:21:17,480 ERROR [140118019524544] (types.h:178) xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range
[ WARN] [1519946477.482304720]: Reconfigure callback failed with exception xioctl(VIDIOC_S_CTRL) failed Last Error: Numerical result out of range:

RealSense-Customer-Engineering, 2018-03-04:
[Realsense Customer Engineering Team Comment] 
Hi @littlejohnyang
My configuration is Ubuntu 14.04.5 + kernel 4.4.0-31 + D410 on the UP board. There is no the warning/error messages.
Suggest to follow below instructions to install the required packages for Ubuntu 14.04 and build the librealsense 2.10.0.
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
Once your librealsense build is done, you can innstall DKMS package without running the kernel patch script.
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
$ echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main' | sudo tee /etc/apt/sources.list.d/realsense-public.list
$ sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
$ sudo apt-get update
$ sudo apt-get install realsense-uvcvideo
$ reboot
After above update, you can run below command to see if uvcvideo driver is updated.
$ modinfo uvcvideo | grep ver
you should see 'version:  1.1.1.realsense-4.4.0'
Make sure above driver version is correct, then the app like rs-capture, realsense-viewer, etc should work normally without your warning/error messages.

hogyadenes, 2018-03-05:
Hi!

I also have a problem regarding this issue. 
During the update process I get an error because my platform (armhf) is not supported by the patch. Is it perhaps in progress or it will not be officially supported?

Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease' doesn't support architecture 'armhf'

RealSense-Customer-Engineering, 2018-03-06:
[Realsense Customer Engineering Team Comment] 
Hi @hogyadenes

So far, the DKMS package doesn't support the ARM platform. which ARM platform do you use?
For Nvidia Jetson platform, need to modify the script "patch-realsense-ubuntu-xenial.sh" to patch.

dorodnic, 2018-03-19:
Hi @littlejohnyang 
Feel free to open a new issue if you need further assistance with patching

closed 2018-03-19 by dorodnic