realsenseai/librealsense #1254: How to set slave mode in D415 camera

sbnn123456, 2018-03-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | librealsense2.0                       | 

### Issue Description
In Intel's paper, Using the RealSense D4xx Depth Sensors in Multi-Camera Configurations:
[image]
We Tried this code below:

	uint8_t input[24] = {0};
	input[0] = 0x14;
	input[2] = 0xab;
	input[3] = 0xcd;
	input[4] = 0x64;
	input[8] = 2;
	auto result = rs2_send_and_receive_raw_data(dev, static_cast<void*>(input), 24, &e);
	check_error(e);

but it still work in default mode(not in slave mode). Any suggestion will be appreciate!

dorodnic, 2018-03-01:
Hi @sbnn123456 
I suspect that the necessary firmware support has not yet been published. Let me check and update this thread on the status. Regarding the code, I believe you are doing everything right.

sbnn123456, 2018-03-01:
@dorodnic  thank you very much.

closed 2018-03-01 by sbnn123456