realsenseai/librealsense #1255: D415 No device connected

liugt34, 2018-03-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415                                                       | 
| Firmware Version                   |05.08.15.00                                              | 
| Operating System & Version |  Windows 10                                          | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC                                                          |
| SDK Version                            |  { legacy / 2.0 }                          | 

the example for C# 
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp/cs-tutorial-2-capture

when i run ,it has an exception

No device connected
rs2_pipeline_start_with_config(pipe:0000000000AFAD10, config:0000000000AFA830)

i have no idea，and it worked well on  Intel.RealSense.Viewer.exe
[image]

closed 2018-03-01 by liugt34

anshulvj, 2018-05-02:
Were you able to get this running? It worked on Windows but not Ubuntu.

liugt34, 2018-05-03:
it worked well now, maybe the driver haven't setup well