realsenseai/librealsense #1258: Control laser with python wrapper

TheMikeyR, 2018-03-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 / D435 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.14.13-041413-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  v2.10.0                          | 
| Language | Python v3.6 | 

### Issue Description
I have some struggle figuring out how to make the correct api calls using the python wrapper, I want to e.g. [control the laser](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser) but when I have to make the check `if (depth_sensor.supports(RS2_OPTION_LASER_POWER))` in python I seem to fail figuring out how to give the correct input. 

I guess I have struggle since there are no suggestion help when using pycharm for other than the basics. Anyone have an example on how to control and check if an info is available using python wrapper? 

I've tried to look through this [file](https://github.com/IntelRealSense/librealsense/blob/5d987c41962ea70b42357abc1d2f4038867445a8/wrappers/python/python.cpp) but it didn't really help me figuring out the approach.

TheMikeyR, 2018-03-01:
I figured it out using `pyrealsense2.option.laser_power` flag, if there are any help to understand the api more when going from c++ docs and write it in python I appreciate.

RealSense-Customer-Engineering, 2018-03-19:
[Realsense Customer Engineering Team Comment] 
The ENUM binding from C++ to Python is actually done in the file you mentioned python.cpp, please refer to the line 37 #define BIND_ENUM which changes the option string get back from C++ function rs2_option_to_string to lower case and replace the space to "_" then map the string to the enum value.
For example, the RS2_OPTION_LASER_POWER return from rs2_option_to_string will be "Laser Power", and then the function will change the "Laser Power" to "laser_power" and link it to the enum value.

Attached the code of controlling the laser power in Python (Simple sample):

pipeline_profile = pipeline.start(config)
device = pipeline_profile.get_device()
depth_sensor = device.query_sensors()[0]
laser_pwr = depth_sensor.get_option(rs.option.laser_power)
print("laser power = ", laser_pwr)
laser_range = depth_sensor.get_option_range(rs.option.laser_power)
print("laser power range = " , laser_range.min , "~",  laser_range.max)
set_laser = 0
#just simply add 10 to test set function
if laser_pwr + 10 > laser_range.max:
 set_laser = laser_range.max
else:
 set_laser = laser_pwr + 10

depth_sensor.set_option(rs.option.laser_power, set_laser)

closed 2018-03-19 by TheMikeyR

AZfrima, 2019-03-07:
> [Realsense Customer Engineering Team Comment]
> The ENUM binding from C++ to Python is actually done in the file you mentioned python.cpp, please refer to the line 37 #define BIND_ENUM which changes [...]

Will this work with the SR300 ?