realsenseai/librealsense #1261: Realsense with GPU

zenving, 2018-03-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |    Linux  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
Hello, I want to know how the process speed can be improved if I use the nvidia jetson TX2. I think the main question is the IO speed, in other word, is memory read speed limited the whole speed, I want to know can I use the cpu in the TX2 to read the data from realsense device, and then use GPU to process the data realtime, dont need memory copy, can I realize that ? Please help me, thx. I need in 15 ms to process one depth frame, to get the information I need, the process include filter, segmentation....

dorodnic, 2018-03-01:
Hi @zenving 

We are actively looking into this question to better support TX2 customers. 
That said, if your goal is HD/Full HD I fear 15ms is extremely challenging to reach. 
From very initial tests, just copying data from / to GPU can take 5-10ms on the Jetson. 

I must admit we are not experts in ARM and GPU programming, but we are planning to make our work public as soon as possible to allow the community to participate.

zenving, 2018-03-01:
@dorodnic Oh, thank you a lot, and can you reply the issue in #1232, PLEASE .

dorodnic, 2018-03-01:
Sure, let us know if you have more questions.

johnjsb, 2018-03-09:
I'm trying to use a realsense d435 with tx2 as well. just wanted to point out the tk1-tx1-tx2 have shared memory model between gpu and cpu. you should check out these 2 examples.
http://arrayfire.com/zero-copy-on-tegra-k1/
http://ohmwardbond.blogspot.com/2017/03/zero-copy-cuda-opencv-and-nvidia-jetson_15.html
the tx2 on a small carrier board and the realsense are ideal together. please intel check into this.

arybalkin, 2018-03-09:
Another user of Jetson TX2 here... would be very interested to see Jetson specific optimization (CUDA or not) added to librealsense.

closed 2018-03-11 by zenving

GustavZ, 2018-03-23:
How can i install librealsense for usin g a D435 on the jetson tx2?