realsenseai/librealsense #1263: Sync frame and understanding frame rates, frame numbers and timestamps

mjscosta, 2018-03-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR300 / D4150 } | 
| Firmware Version                   | 3.21.0.0 (SR300)/ | 
| Operating System & Version |   Linux (Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.4.0-116-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.10.0            | 

### Issue Description
I've captured frame data, hardware timestamps, time of arrival timestamps and frame numbers, and something seems strange. While I've configured the camera with the following settings:
```
        cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_RGB8, 30);
        cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
        ...
        pipe_.start(config)
```
The frame numbers and timestamps from the depth and color frames, seem to show that the rgb frame rate its almost 1/4 (~5 fps) of the depth sensor, could that be right (check the data bellow)?

Is it possible that I'm doing something wrong? This is the way I'm collecting the data:
```
		rs2::frameset data = pipe_.wait_for_frames();

		rs2::video_frame color = data.get_color_frame();
		rs2::depth_frame depth = data.get_depth_frame();

                print_frame_info(color, *log_file);
                print_frame_info(depth, *log_file);
                (*log_file) << std::endl;

```

stream type | hw ts | librs2 ts | frame nr | stream type | hw ts | librs2 ts | frame nr | delta hw ts | abs delta hw ts
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
Color | 23196.4 | 1519905985135 | 0 | Depth | 23224.8 | 1519905985155 | 17 | -28.4 | 28.4
Color | 23196.4 | 1519905985135 | 0 | Depth | 23258.2 | 1519905985189 | 18 | -61.8 | 61.8
Color | 23196.4 | 1519905985135 | 0 | Depth | 23291.6 | 1519905985222 | 19 | -95.2 | 95.2
Color | 23196.4 | 1519905985135 | 0 | Depth | 23325 | 1519905985255 | 20 | -128.6 | 128.6
Color | 23399 | 1519905985337 | 1 | Depth | 23325 | 1519905985255 | 20 | 74 | 74
Color | 23399 | 1519905985337 | 1 | Depth | 23458.6 | 1519905985389 | 24 | -59.6 | 59.6
Color | 23399 | 1519905985337 | 1 | Depth | 23492 | 1519905985422 | 25 | -93 | 93
Color | 23399 | 1519905985337 | 1 | Depth | 23525.4 | 1519905985456 | 26 | -126.4 | 126.4
Color | 23601.5 | 1519905985540 | 2 | Depth | 23525.4 | 1519905985456 | 26 | 76.1 | 76.1
Color | 23601.5 | 1519905985540 | 2 | Depth | 23659 | 1519905985589 | 30 | -57.5 | 57.5
Color | 23601.5 | 1519905985540 | 2 | Depth | 23692.4 | 1519905985623 | 31 | -90.9 | 90.9
Color | 23601.5 | 1519905985540 | 2 | Depth | 23725.8 | 1519905985656 | 32 | -124.3 | 124.3
Color | 23804.1 | 1519905985742 | 3 | Depth | 23725.8 | 1519905985656 | 32 | 78.3 | 78.3
Color | 23804.1 | 1519905985742 | 3 | Depth | 23892.9 | 1519905985823 | 37 | -88.8 | 88.8
Color | 23804.1 | 1519905985742 | 3 | Depth | 23926.3 | 1519905985857 | 38 | -122.2 | 122.2
Color | 23804.1 | 1519905985742 | 3 | Depth | 23959.7 | 1519905985890 | 39 | -155.6 | 155.6
Color | 24006.7 | 1519905985945 | 4 | Depth | 23959.7 | 1519905985890 | 39 | 47 | 47
Color | 24006.7 | 1519905985945 | 4 | Depth | 24059.9 | 1519905985990 | 42 | -53.2 | 53.2
Color | 24006.7 | 1519905985945 | 4 | Depth | 24093.3 | 1519905986024 | 43 | -86.6 | 86.6
Color | 24006.7 | 1519905985945 | 4 | Depth | 24126.7 | 1519905986057 | 44 | -120 | 120

aangerma, 2018-03-04:
Hi @mjscosta can you please validate your actual fps by running 'realsense-viewer' and start streaming the RGB camera, then push the stream info button (in the top right corner),
in the text box that will show up you can see the FPS.

mjscosta, 2018-03-04:
Hi @aangerma , I confirm the fps, but only with the SR300 camera. D415, does not seem to have this  problem.

[image]

mjscosta, 2018-03-05:
We figured the problem out, by comparing with another camera. As it seems was the auto-exposure disabled, caused this behaviour in both cameras.

aangerma, 2018-03-07:
The problem is probably that the RGB exposure value was high so the FPS of the color decreased,
you can validate it by clicking the 'Controls' to see all the camera controls and epically the exposure (when it manual).

mjscosta, 2018-03-08:
Right, thanks, I'm reading the datasheet, and now clearly understand the issue.

closed 2018-03-19 by dorodnic