realsenseai/librealsense #1266: The different between D435 and D415

zenving, 2018-03-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
I know the D435 is global shutter, and the D415 is rolling shutter, but I want to know if all the 3 cameras in D415 is rolling shutter, and all the 3 cameras in D435 is global shutter ? In other word, if I want  to track the moving thing, which speed is 10m/s, and I only need to use depth frame to capture it, should I buy a new D435 ? Because I dont know the different type of shutter is only influence the color frame or influence the depth frame too.Please help me, thx.

zivsha, 2018-03-04:
Hi @zenving ,
You can find this information in the [D400 technical datasheet](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-D400-Series-Datasheet.pdf).

RealSense-Customer-Engineering, 2018-03-09:
[Realsense Customer Engineering Team Comment] 
hi @zenving,

would like to know if you find anything you want?
thanks.

closed 2018-03-11 by zenving