realsenseai/librealsense #1267: [Question] What is "Depth scale calibration" mean?

unagiHuman, 2018-03-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
I read the following manual, but I did not understand the meaning of "Depth scale calibration".

https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_D400_Dyn_Calib_User_Guide.pdf

I understand that "Depth scale calibration" is an act of adjusting the distance from the camera that depth can be acquired with high accuracy, but do you understand it well?

zenving, 2018-03-02:
I think the real distance value in meter is the raw value in every raw pixel data from the depth frame 
multipy the scale.

RealSense-Customer-Engineering, 2018-03-06:
[Realsense Customer Engineering Team Comment] 
Hi @unagiHuman
As you can check the doc, the depth scale calibration requires a targeted chart and it aligns the depth frame due to changes in position of the optical elements. The process checks for rectification error (absolute Y difference) but also compares the measured pattern size with ground truth. The calibration process is to mainly optimize the extrinsic parameters and help recover the depth accuracy if the realsense camera has something wrong, ex: bending, drop, etc.

RealSense-Customer-Engineering, 2018-03-09:
[Realsense Customer Engineering Team Comment] 
HI @unagiHuman,

The calibration can correct the deviation/rotation of the camera sensor position. Each D415/D435 has the factory calibration to calibrate the intrinsic and extrinsic parameters. The dynamic calibration for end users can help calibrate the extrinsic parameters like translation and rotation between two cameras in D415/D435.
The maximum use range depends on test scene, light condition, IR laser power, etc. Indeed, the depth accuracy will be impacted/changed if a far distance used. More texture on the target surface can help the depth. Or, IR pattern (from IR projector) can help the depth.

unagiHuman, 2018-03-13:
Hi @RealSense-Customer-Engineering 

I understood raising the IR laser power and trying it in a weak dark room of light in order to improve the depth accuracy of the distance far.

But I could not understand the following sentences you say.
> Or, IR pattern (from IR projector) can help the depth.
Increasing the power of IR laser is possible by setting Realsense parameters, but is it possible to control the IR pattern?

agrunnet, 2018-03-13:
The projected IR pattern cannot be controlled or modified electronically, apart from simply adjusting the power. However, using external projectors is fine as long as it
1. Is on either DC (continuously) or at >50Khz. 
2. If it uses lasers all care should be taken to reduce laser speckle. One option is to move the projector for example. This can actually be beneficial and help reduce the depth ripple a bit if you use averaging as well.
3. It is eye safe (of course)
4. Wavelength is between 400 and 1000nm
5. Pattern should not be periodic, but semi-random. A pattern of between 5000 and 300000 dots will work. 
6. No extra calibration is needed by the RealSense camera. It does not care about the nature of the projector. 
7. Multiple projectors can be used.  This will help with occlusions.  There are no real constraints otherwise on where you place the projectors. 
8. The projectors do not have to be on same stiffener as the RealSense camera.

unagiHuman, 2018-03-13:
Hi @agrunnet 

I read your answer and it glows.
When preparing two RealSense and irradiating the IR projector on the same surface, I thought that the accuracy of the depth to be acquired would be high.

agrunnet, 2018-03-13:
Yes, you are right. 
In general adding more texture will help depth. 
There are also other factors to consider, namely that a well textured passive target actually gives best results. 
Here is an example target. 
[image]
Why does it give better results? 
1. There is high spatial frequency content 
2. There is no laser speckle. We found that even after all the precautions taken in to reduce speckle in the D415 and D435, using a passive target can give 30% less depth noise.  

So the upshot is that in some cases it may actually be better to turn off the projector and simply illuminate well with white light. The RealSense cameras actually behave quite well outside in bright sunlight because the natural texture becomes even more visible. The stereo depth will always be limited by the quality of the left and right images. It is well known from cell phone cameras, for example, that they give better images in daylight than in dimly lit indoor environments. The same applies to the small RealSense cameras.

unagiHuman, 2018-03-14:
Hi @agrunnet
Thank you for your important findings.
I was surprised that Realsense can recognize the above image with the Realsense IR camera (RGB?) And get the depth. And without the pattern of IR laser.
I will try to see if you can get the depth of the distance accurately in several ways you taught.

dmitryshendryk, 2018-05-16:
Found it

RealSense-Customer-Engineering, 2018-06-04:
[Realsense Customer Engineering Team Comment]
Hi @dmitryshendryk,

Do you get the better results with @agrunnet suggestions?

RealSense-Customer-Engineering, 2018-07-09:
[Realsense Customer Engineering Team Comment]
Hi @dmitryshendryk,

any update there?

closed 2018-07-23 by dorodnic