realsenseai/librealsense #1268: Firmware update tool for linux

TheMikeyR, 2018-03-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  4.14.13-041413-generic                                      | 
| Platform                                 | PC / Nvidia Jetson  |
| SDK Version                            |  v2.10.0                       | 

### Issue Description
Any plans on making a firmware update tool for linux, currently only [windows](https://downloadcenter.intel.com/download/27514/Windows-Device-Firmware-Update-Tool-for-Intel-RealSense-D400-Product-Family) is available.

jlgarcia75, 2018-03-02:
Hi @TheMikeyR,

The FW update tool for Linux is in development and will be released relatively soon. Sorry can't give you a solid timeline.

Regards,
Jesus G.
Intel Customer Support

TheMikeyR, 2018-03-05:
Thanks @jlgarcia75 looking forward for the release! Should I close the issue or keep it open until the tool have released?

RealSense-Customer-Engineering, 2018-04-13:
[Realsense Customer Engineering Team Comment] 
Hello @TheMikeyR,

I will close this issue. Please monitor this page, https://realsense.intel.com/intel-realsense-downloads/, for tool releases. The Linux FW update tool will be released soon.

Regards,
Jesus G.
Intel Customer Support

closed 2018-04-13 by RealSense-Customer-Engineering

poripsa, 2018-05-11:
Hi, on running 

echo 'deb http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial main'| sudo tee /etc/apt/sources.list.d/realsense-public.list 

as the above document suggests and the running 

sudo apt-get update 

Results in:
W: The repository 'http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo xenial Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.

E: Failed to fetch http://realsense-hwpublic.s3.amazonaws.com/Debian/apt-repo/dists/xenial/main/binary-amd64/Packages  404  Not Found

and then:

E: Couldn't find any package by glob 'intel-realsense-dfu*'
E: Couldn't find any package by regex 'intel-realsense-dfu*'

when trying to install package. I am on Ubuntu 16.04 trying to flash the latest firmware in D415 camera.
Let me know further steps to take.

TheMikeyR, 2018-05-14:
@poripsa it is only possible to flash it with a windows host, the linux version will be made available later.

hello-sea, 2019-03-29:
@poripsa

hello-sea, 2019-03-29:
@poripsa Hello, I found the cause of the problem and hope to help you.
Please refer to:
[1] https://github.com/IntelRealSense/librealsense/issues/2170#issuecomment-409150769