realsenseai/librealsense #1269: Difference in the modules? Availability etc.

TheMikeyR, 2018-03-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 / T150 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  4.14.13-041413-generic                                      | 
| Platform                                 | PC / Nvidia Jetson  |
| SDK Version                            |  v2.10.0                       | 

### Issue Description
Currently there is an issue going on which also answer one of my question regarding main differences https://github.com/IntelRealSense/librealsense/issues/1266 but furthermore I'm in doubt of the following:

1. D410 and D415 only difference is an added color sensor to the "camera-module" ?
2. D430 and D435 only difference is an added color sensor to the "camera-module" ?
3. Will D415 and/or D435 camera module be available for buy? (without casing etc. only module, just like D410 etc.)

     3a. Is there any expectations on shipping date for the camera-modules alone?
4. What is the point of T150, is it used to know the world positioning of the D400-series cameras? it's mentioned in the [datasheet](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-D400-Series-Datasheet.pdf) but I can't seem to find any more info on purpose, shipping, cost etc.

jlgarcia75, 2018-03-02:
Hello @TheMikeyR,

1. D410 and D415 only difference is an added color sensor to the "camera-module"? YES
2. D430 and D435 only difference is an added color sensor to the "camera-module"? YES
3. I will find out for you.
4. The purpose of the T150 tracking module is to add motion tracking capabilities to enable SLAM use cases - very useful in robotics. It has an embedded IMU (gyroscope, accelerometer), for movement and rotation detection, and fisheye camera. I will find out about the availability of this part for you too.

Regards,
Jesus G.
Intel Customer Support

RealSense-Customer-Engineering, 2018-03-09:
[Realsense Customer Engineering Team Comment] 
hi @TheMikeyR,,
we have D415 on Design In Tools as link below:
https://designintools.intel.com/SearchResults.asp?Cat=2358
FYR.

RealSense-Customer-Engineering, 2018-03-09:
[Realsense Customer Engineering Team Comment] 
sorry @TheMikeyR,
you may have no access to Design In tools, if this the case, please goto
https://click.intel.com/realsense.html
for D415/D435 cameras.

RealSense-Customer-Engineering, 2018-03-20:
[Realsense Customer Engineering Team Comment] 
hi @TheMikeyR,

do you still need support on this topic?
thanks.

TheMikeyR, 2018-03-20:
Hi @RealSense-Customer-Engineering yes, I'm still awaiting @jlgarcia75 answer on question 3 and 3a.

RealSense-Customer-Engineering, 2018-03-20:
[Realsense Customer Engineering Team Comment] 
Hello @TheMikeyR,

I am sorry it has taken so long for me to respond. There are currently no plans to sell the D415 and D435 cameras as bare modules.

Regarding question 4 on the T150, it is not available for purchase to the public and currently, there are no plans to make it so.

Regards,
Jesus
Intel Customer Support

closed 2018-03-21 by TheMikeyR