realsenseai/librealsense #1270: Can't get PLY file.

teruterubozu808, 2018-03-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 05.08.15.00| 
| Operating System & Version |  MacOS 10.13.3  | 
| Platform                                 | PC|
| SDK Version                            | v2.10.0| 

### Issue Description
I have a quiestion about realsense-viewer.

In realsense-viewer, I found a floppy image icon in 3D mode, which shows, "Export 3D model to PLY format". I clicked the icon but nothing happened(no output PLY file produced). 

Is this function available in realsense-viewer?

zivsha, 2018-03-04:
Hi @teruterubozu808 ,

This is a [known issue for MacOS ](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#known-issues-on-mac-os) - Open/Save file dialog windows are not working. This will be the same if you choose "Add Source"-->"Load Recorded Sequence".
The only workaround I can suggest, is calling the [`export_to_ply` function](https://github.com/IntelRealSense/librealsense/blob/9437bc24c0f4ce16193f1045ceea703a8151b96d/include/librealsense2/hpp/rs_frame.hpp#L513) from your code. A python example for this is available [here](https://github.com/IntelRealSense/librealsense/blob/c8ee8fa1912b9297df13bfe097d527667fe0afba/wrappers/python/examples/export_ply_example.py)

teruterubozu808, 2018-03-04:
Hi @zivsha,

Thank you for your advice. 
So, it is impossible to save PLY files by using GUI menu, I understand.

I will set up pyrealsense2 and try to use the informed python program.
(I will update this issue after trying above.)

zivsha, 2018-03-04:
It is impossible on Mac at the moment.
You do not have to use python, you can do it with C/C++ as well

teruterubozu808, 2018-03-04:
Thank you for your quick reply.
I will try it by using C/C++.

teruterubozu808, 2018-03-06:
I inserted `points.export_to_ply("out.ply", frames);` below [this line](https://github.com/IntelRealSense/librealsense/blob/5121ab030c939632d4d4a36ac999a937c3b2e170/examples/pointcloud/rs-pointcloud.cpp#L44),
 built it and excuted `rs-pointcloud`.

Then I succeeded in acquiring a PLY file!

(I checked the PLY file using "Meshlab" and found it a little bit strange. 
 It seems that timing adjustment or some other treatments are needed. )

Anyway, I got a PLY file.
Thanks a lot !

closed 2018-03-06 by teruterubozu808

hdjsjyl, 2018-03-06:
@teruterubozu808 , @zivsha  Hi, thanks for your experience. I am using D435 and Ubuntu. I can get .ply file from GUI. But when I use meshlab to open it, it shows a point cloud of tripod of camera rather than people. So do you know the reason?

zivsha, 2018-03-06:
@hdjsjyl maybe you need to rotate the view...

teruterubozu808, 2018-03-07:
Hi,@hdjsjyl 

In my case, the original axis orientation of the Meshlab pointcloud window  was as follows. 

 x:right
 y:top
 z:backward

And I had to rotate the view as follows to get the correct looks. 

 x:right
 y:down
 z:forward

To show the axes, click the icon which shows, “Draw XYZ axes in world coordinates ”.
You can rotate the view by using your mouse, but I think you also need “Manipulators Tool” to adjust the figure’s position. (The icon of “the Manipulators Tool” is also placed on the top of the pointcloud window.)

tommyboy660, 2018-06-29:
HI I recently got my D435 Camera unit delivered to me about 2 weeks ago! been playing around a bit with it & i'm a basic beginner when it comes to coding by way of freehand/ ie know code to type out of my head! I mean if I have something basic to start from (ie code) & then at least something in the way of a sample code book that may show what each code word or string defines or something to that effect! I've read heavily into this module of realsense devices as well as the few apps & apk that are already compatible for use with it & from my understanding the viewer I thought was supposed to be able to export the scanned object etc. in ply format which as i've read gave me the impression that it was able to export a scanned session or ie more like a video that you get from a video camera rather then just a screenshot type file!!
How do i go about getting or I guess Repack many ply files into a single file as to then import into another app so that it either automatically recognizes it as a point cloud of the entire object or at least allow me to have enough to work with that I can make it all into a point cloud or straight to a mesh?? Also what software apps do you all recommend using 1st: for the coding 2nd: the best ones to use that either are able to capture straight from the realsense devices or that are able to work the best with the output files??? Also with regarding the previous two questions not only the best software apps but also the most user friendly/ easiest to learn?? 

I already have most all the Autodesk apps, Meshlab, meshmixer, blender, 3DF Zephyr along with various other apps! 

Any help & assistance is deeply appreciated, thank you all for anything & everything that you can hep me with & share with me. Tom Card

bingws, 2019-04-28:
> It is impossible on Mac at the moment.
> You do not have to use python, you can do it with C/C++ as well

hi , gay. Look at [this](https://github.com/IntelRealSense/librealsense/issues/3859), how to change the python program, do not generate binary files, but generate readable files?