realsenseai/librealsense #1271: rs2_get_video_stream_intrinsics returns wrong FOV for D435

SirDifferential, 2018-03-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   Debian 9 | 
| Kernel Version (Linux Only)    | 4.9.65 | 
| Platform                                 | PC  |
| SDK Version                            |  latest master from git | 

### Issue Description

Querying depth stream FOV using `rs2_get_video_stream_intrinsics` with D435 returns incorrect FOV values: 69.681961 x 42.772346. The correct FOV should be 91.2 x 65.5. D415 returns the correct FOV.

See: https://github.com/SirDifferential/minimal_realsense2/blob/master/main.c#L583

agrunnet, 2018-03-02:
It it possible it is reading the third color camera? This one has that FOV in the D435.

zivsha, 2018-03-04:
Hi @SirDifferential ,
Which stream's intrinsics are you taking?
For simplicity you can run `rs-enumarate-devices.exe -c` and share the results.

SirDifferential, 2018-03-05:
My mistake:

```
rs2_config_enable_stream(config, RS2_STREAM_DEPTH, -1, cDepthW, cDepthH, RS2_FORMAT_Z16, 30, &e);
rs2_config_enable_stream(config, RS2_STREAM_COLOR, -1, cColorW, cColorH, RS2_FORMAT_RGB8, 30, &e);
selection = rs2_config_resolve(config, pipe, &e);
stream_list = rs2_pipeline_profile_get_streams(selection, &e);
stream_list_count = rs2_get_stream_profiles_count(stream_list, &e);
rs2_pipeline_start_with_config(pipe, config, &e);

for (stream = 0; stream < rs_state->stream_list_count; stream++) {
    rs2_stream str;
    rs2_format format;
    int index;
    int id;
    int fps;

    const rs2_stream_profile* prof = rs2_get_stream_profile(stream_list, stream, &e);
    rs2_get_stream_profile_data(prof, &str, &format, &index, &id, &fps, &e);

    rs2_intrinsics intrinsics;
    if (stream == RS2_STREAM_DEPTH)
    {   
        rs2_get_video_stream_intrinsics(prof, &intrinsics, &e);
        rs2_fov(&intrinsics, fov);
    }   
    else if (stream == RS2_STREAM_COLOR)
    {   
        rs2_get_video_stream_intrinsics(prof, &intrinsics, &e);
        rs2_fov(&intrinsics, rgb_fov);
    }   
}
```

There's a mistake in this where I incorrectly checked the `stream` variable in the for loop instead of the `str` which was assigned by the `rs2_get_stream_profile_data` call. Changing this correctly returns the device FOV.

closed 2018-03-05 by SirDifferential