realsenseai/librealsense #1272: Incorrect depth estimation

maciekd, 2018-03-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.08.09.00 | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.4.0-116-generic | 
| Platform                                 | Intel NUC  |
| SDK Version                            | 2.10.0 }                          | 

We're running a bunch of RS400 sensors and in our test lab we have a repetitive pattern on the walls. When the wall is out of distance for the laser projector we get incorrect depth readings sometimes.

Here is depth data:
https://www.dropbox.com/s/z6k7t6i8x847wqv/Screen%20Shot%202018-03-02%20at%2016.37.03.png?dl=0

Color data:
https://www.dropbox.com/s/4i1y5uns5a6uuhl/Screen%20Shot%202018-03-02%20at%2016.37.32.png?dl=0

Are there any parameters we can tweak in order to minimize the amount of false readings?

Thanks in advance.

zivsha, 2018-03-04:
Hi @maciekd , sorry but we're going to need one of you drones to help...
Just kidding 😄 
You can start by using one of the [pre-defined presets](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets) (also see links at the buttom), and continue with tweaking the advanced controls to find the best matching configuration for your use case. 
Unfortunately, to my best knowledge, we do not have any documentation for each advanced control setting, but feel free to ask about them here and we will try to help

maciekd, 2018-03-05:
Fantastic, will ha a look at this, thanks!

closed 2018-03-05 by maciekd