realsenseai/librealsense #1273: SR300 no frames in Unity

CDragu, 2018-03-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 (in a Razer Stargazer)                  | 
| Firmware Version                   | 3.21.0.0                                                   | 
| Operating System & Version |   Win 10                                                  | 
| Platform                                 | PC                                                           |
| SDK Version                            |  2.10.1                                                    | 

I am getting error rs2_keep_frame in Unity when I try to preview the depth sensor stream. The box for depth preview stays white and the application freezes if Process Mode is set to Unity Thread. 

No other preview is working in the application(eg. colour, IR, Point Cloud, etc.)

Here is the full log from Unity:
`rs2_keep_frame
UnityEngine.Debug:LogError(Object)
RealSenseDevice:ProcessFrames(FrameSet) (at Assets/RealSenseSDK2.0/Scripts/RealSenseDevice.cs:186)
RealSenseDevice:Worker_DoWork(Object, DoWorkEventArgs) (at Assets/RealSenseSDK2.0/Scripts/RealSenseDevice.cs:200)
System.ComponentModel.BackgroundWorker:ProcessWorker(Object, AsyncOperation, SendOrPostCallback)
`

zivsha, 2018-03-04:
Hi @CDragu 
Does this happen right on start or after a while? 
Are you able to use the camera with the RealSense Viewer / run the C# samples?

CDragu, 2018-03-04:
Hi,
Happens right on start, no frames go trough.
The RealSense Viewer and the old SDK are working but I am not able to run the C# wrapper. 
With Visual Studio set to realease mode, 86X and trying to run any of the examples I get:

`Unhandled Exception: System.EntryPointNotFoundException: Unable to find an entry point named 'rs2_keep_frame' in DLL 'realsense2'.
   at Intel.RealSense.NativeMethods.rs2_keep_frame(IntPtr frame)
   at Intel.RealSense.Frame..ctor(IntPtr ptr)
   at Intel.RealSense.FrameSet.CreateFrame(IntPtr ptr)
   at Intel.RealSense.FrameQueue.WaitForFrame()
   at Intel.RealSense.Program.Main(String[] args) in C:\Users\cosmi\Desktop\librealsense-development\librealsense-development\wrappers\csharp\cs-tutorial-1-depth\Program.cs:line 45
Press any key to continue . . .`

More information: 

- RealSense Device Inspector fills with information at the start, including serial number and name of the camera.

- If I start the default scene I get error `Exception: Failed to resolve request. No device found that satisfies all requirements` and removing the Infrared profile allows it to go over that error.

zivsha, 2018-03-04:
It looks like you are using an old version of `realsense2.dll` (the native dll of the library which is used by the csharp wrapper).
Can you please make sure you build the C# wrapper using the matching native dll version (either build both from the same branch or use both from the installer - make sure you don't mix different versions of the installer).

CDragu, 2018-03-04:
Thanks for the help, finally figured it out. I must have got confused in the versions I had downloaded, also I couldn't compile a 64bit version of the .dll. Now I downloaded the RealSense SDK 2.0 installer and used the .dll files from there and everything seems to be working.

closed 2018-03-04 by CDragu