realsenseai/librealsense #1275: `examples/align` crash with D435 on macOS: Frame of requested stream type was not found!

moorage, 2018-03-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version |   macOS 10.13.3  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | direct connect via USB  |
| SDK Version                            |  2.10.0                         | 

### Issue Description

I can successfully compile both `librealsense2` and `examples/align` at https://github.com/IntelRealSense/librealsense/tree/master/examples/align  

However, when I run the rs-align binary, I immediate get a whole bunch of output with the final one being 

```
$ ./rs-align 
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
unsupported descriptor subtype: 13
RealSense error calling ():
    Frame of requested stream type was not found!
```

I can run `realsense-viewer` without a crash, however, I do constantly get the `unsupported descriptor subtype: 13` message.

[image]

ahpuck, 2018-04-05:
What release is this fix targeting? I'm also having this issue with version 2.10.2.

RealSense-Customer-Engineering, 2018-04-17:
[Realsense Customer Engineering Team Comment] 
hi @moorage, @ahpuck,

would like to confirm if v2.10.3 release solve your issue?
thanks.

ahpuck, 2018-04-19:
rs-align is working now though with any reasonable movement it crashes with this error. I just tested with v2.10.4. 

RealSense error calling ():
    Frame of requested stream type was not found!

JasonChang13, 2018-04-25:
hi @ahpuck,

I've tested the rs-align of libRealsense v2.10.4, it's working good, can you please tell what reasonable movement means? moving the camera around?
please let us know how to reproduce your issue, thanks.

Regards,
Jason
Intel RSG Customer Platform Engineering

RealSense-Customer-Engineering, 2018-05-21:
[Realsense Customer Engineering Team Comment]
hi @ahpuck,

do you still need help on this topic?
thanks.

ahpuck, 2018-05-24:
Nope, I think I'm all sorted for now. Updating the application seemed to
fix that issue for the most part.
Austin Puckett
(970) 231-4364
austinhpuckett@gmail.com

closed 2018-05-25 by RealSense-Customer-Engineering