realsenseai/librealsense #1276: Cannot get the public key for installing realsense even used a new key

polpolpol01, 2018-03-05:
### Issue Description
I tried to follow the instruction provided in ReadMe.md

[image]

However, after used command sudo apt-get key as shown in below figure

[image]

I cannot get the key and when I sudo apt-get update. The computer showed that i did not get the key to access downloading realsense. Do you have any suggestion about this problem?

ev-mp, 2018-03-05:
Hello @polpolpol01 ,
The actual error  is that your PC could not reach GPG server to obtain the public key:
> .. gpg: keyserver timed out

[Googling this error up](https://unix.stackexchange.com/questions/75892/keyserver-timed-out-when-trying-to-add-a-gpg-public-key) suggests that it might be a firewall getting in the way.
Please try the methods recommended and update so we will adjust the troubleshooting accordingly.

polpolpol01, 2018-03-05:
@ev-mp Thank you for your suggestion. I already googled and now found the solution to avoid the firewall and get the key. Here is the solution https://lodge.glasgownet.com/2012/07/11/apt-key-from-behind-a-firewall/

closed 2018-03-05 by polpolpol01

ev-mp, 2018-03-05:
Thanks, we will add your solution to Troubleshooting