realsenseai/librealsense #1279: Camera model: D415, D435

AleBasso80, 2018-03-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

AleBasso80, 2018-03-05:
Camera Model: D400
Firmware Version: 05.08.15.00
Operating System & Version: Linux Ubuntu 16.04
Kernel Version (Linux Only):  4.13.0
Platform: PC, NVIDIA Jetson.
SDK version: 2.10.0

Hello,
I tested both D415 and D435 with PC and Jetson TX2.
After some regular tests made both on a PC and on a Jetson TX2, neither my C++ program nor the realsense-viewer are any more able to detect the RGB module. Moreover, the devices are no more recognized as D415 or D435, but as generic IntelRealSense USB2.
I don't know why this is happening nor how to restore the original firmware.

I configure the streams by means of the command configuration.enable_stream and I set the preset as follows:
  auto range = depth_sensor.get_option_range(RS2_OPTION_VISUAL_PRESET);
  for (auto i = range.min; i < range.max; i += range.step)
    if (std::string(depth_sensor.get_option_value_description(RS2_OPTION_VISUAL_PRESET, i)) == _preset)
      depth_sensor.set_option(RS2_OPTION_VISUAL_PRESET, i);

I picked this code up from an example. 
These are the only parts of my code where I set something to the camera.

Any idea?

Thanks!

ev-mp, 2018-03-05:
Hello @AleBasso80 ,
The root cause is the improper enumeration of the device as USB2 type. D400 devices require USB3 connection to satisfy both the power and bandwidth requirements. 
When the device is enumerated as USB2 the firmware switches to a "reduced capability" mode which includes only a subset of depth resolutions and formats (only low-res) and no RGB sensor.

The main reason for the improper enumeration is the HW setup: USB ports/connectors/cables/hubs.
You need to verify the HW chain step by step to fix it.

Some recent reports on this topic: [1181](https://github.com/IntelRealSense/librealsense/issues/1181),  [1061](https://github.com/IntelRealSense/librealsense/issues/1061),  [1148](https://github.com/IntelRealSense/librealsense/issues/1148),  [902](https://github.com/IntelRealSense/librealsense/issues/902)

AleBasso80, 2018-03-05:
Hello @ev-mp,
thank you for your answer. Actually I plugged the cameras on the wrong USB port on the PC, but the Jetson's was correct. I completely solved the issue by upgrading the firmware to the latest version.
I still have a concern: when I try to set the preset 9 times out of 10 an exception is thrown. As I wrote, I use the procedure suggested within an example. I need to solve it and I'm also wondering if keeping on trying to set a new preset unsuccessfully might lead to some internal camera issues.
Do you have any idea?
Cheers

ev-mp, 2018-03-13:
Hello @AleBasso80 , is it reproducible by switching the Presets in the `realsense-viewer` application ?

AleBasso80, 2018-03-16:
Hello @ev-mp , in the realsense-viewer it works. I was wondering whether I had made some mistakes, but I just followed the procedure shown in an example.

RealSense-Customer-Engineering, 2018-04-16:
[Realsense Customer Engineering Team Comment] 
@AleBasso80
Problem fixed? Can it be closed?

closed 2018-07-26 by dorodnic