realsenseai/librealsense #1280: Depth Values Fluctuation

vladsterz, 2018-03-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 and D435  | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |   2.9.0                         | 

### Issue Description
So I am messing around with the sensor and I noticed (I guess not the only one) the fluctuating noise. Here is a snapshot of a wall.

[image]

Has anyone found any combination of depth device configuration to minimize it?

If not, has anyone found any work/research on this problem?
Thanks in advance!

zivsha, 2018-03-05:
Hi @vladsterz , 
Please check out our Post Processing example and the RealSense Viewer's filters (they both use the same utility) to see how to reduce noise, and improve depth images. In addition you might find our [pre defined presets](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets) useful

vladsterz, 2018-03-05:
Hello @zivsha ,
I've already checked presets, with high accuracy giving the best results. On the other hand the pointcloud is extremely sparse (for not planar objects especially).  
Will definetly check the Post Processing example. Is there anyway that some utilities are not implemented in my SDK version?

zivsha, 2018-03-05:
We consistenly add features and improvements so yes it is possible, thought I'm not sure why you are asking this. Anyway, you can see the changes in our [release notes](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes)

RealSense-Customer-Engineering, 2018-04-09:
[Realsense Customer Engineering Team Comment] 
@vladsterzI
Does the problem still exist?

vladsterz, 2018-04-09:
Has anything changed?

RealSense-Customer-Engineering, 2018-04-10:
[Realsense Customer Engineering Team Comment] 
New 2.10.2 has released, improved the post processing. BTW, the problem still exist after you apply the post processing?

Can you attach the IR image? just want to make sure the camera sees clear image to generate good enough depth.
Also can you let me know how far the target away from camera?

bochen87, 2018-04-23:
The post processing makes it a bit better, but unfortunately, it's still very far away from the performance of a Kinect.

dorodnic, 2018-07-12:
We published dedicated whitepaper on this subject:
[Best-Known-Methods for Tuning Intel® RealSense™ D400 Depth Cameras for
Best Performance](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/BKMs_Tuning_RealSense_D4xx_Cam.pdf)

closed 2018-07-12 by dorodnic